mavlink-router/mavlink-router #22: Connect to ardupilot sitl?

fnoop, 2017-02-22:
Is it possible to connect mavlink-router to ardupilot sitl?  It looks like sitl only outputs on tcp (5560 by default), and I can't see how to specify tcp address to mavlink-router, just serial or udp.

edersondisouza, 2017-02-22:
Currently, mavlink-router only connects to UDP or serial - not TCP directly. You could use MAVProxy to expose Ardupilot SITL via UDP - as [seen here](http://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html#sitl-with-mavproxy-udp) - then connect mavlink-router to it - not sure if that's what you're looking for, though.

fnoop, 2017-02-22:
@edersondisouza No I'm looking to replace mavproxy, it's a bit cpu-intensive just for mavlink proxy.  Is it relatively simple to adapt mavlink-router to connect to tcp?

edersondisouza, 2017-02-22:
Yes, it actually should be. It would be a matter of a new `Endpoint` class - much like `TcpEndpoint` which currently is a 'server', it would have to be extended to act as a client - and some bits to wire it into current code.

lucasdemarchi, 2017-02-28:
@edersondisouza could you take a look into this?  Maybe it's also better to change the configuration format so we have:

```
[TcpEndpoint  bla]
...

[UdpEndpoint  bli]
...

[UartEndpoint blu]
..
```

Since each of them has different options.  On TcpEndpoint we could have a flag `Type = server | client`. Thoughts?

edersondisouza, 2017-03-01:
It's a better format than our current one - it makes easier to spot differences. I'm assuming that we can have N client `TcpEdpoint`s, right?

fnoop, 2017-03-15:
Confirmed working.  Thanks very much :)))

closed 2017-03-15 by fnoop