mavlink-router/mavlink-router #25: Output spam - No message entry

fnoop, 2017-02-24:
```
Feb 23 23:58:39 fnoop-raspberry mavlinkrouter.sh[9455]: (endpoint.cpp:269) No message entry for 178
Feb 23 23:58:39 fnoop-raspberry mavlinkrouter.sh[9455]: (endpoint.cpp:269) No message entry for 182
Feb 23 23:58:39 fnoop-raspberry mavlinkrouter.sh[9455]: (endpoint.cpp:269) No message entry for 163
Feb 23 23:58:39 fnoop-raspberry mavlinkrouter.sh[9455]: (endpoint.cpp:269) No message entry for 165
Feb 23 23:58:39 fnoop-raspberry mavlinkrouter.sh[9455]: (endpoint.cpp:269) No message entry for 173
Feb 23 23:58:39 fnoop-raspberry mavlinkrouter.sh[9455]: (endpoint.cpp:269) No message entry for 193
Feb 23 23:58:39 fnoop-raspberry mavlinkrouter.sh[9455]: (endpoint.cpp:269) No message entry for 178
Feb 23 23:58:39 fnoop-raspberry mavlinkrouter.sh[9455]: (endpoint.cpp:269) No message entry for 182
```

Looks like is when mavlink-router can't look up the incoming message against mavlink library?  Not sure why this is the case, but the output should at least be an option or rate-limited, this is killing the system journal logging in this case, as it's launched from systemd.

lucasdemarchi, 2017-02-24:
Yep... this message was recently added and should be fixed.

For the unknown messages, it's probably related to using common.xml to generate the headers. You are using ardupilot, right?  We need to have an option to use another xml.

fnoop, 2017-02-24:
Yes exactly, using ardupilot (copter 3.5-dev/master).

lucasdemarchi, 2017-02-28:
@fnoop the excessive logging is being fixed by #27. We will add another patch to allow to use the ardupilot xml instead of common.xml.

fnoop, 2017-03-15:
Fixed, thanks very much :)))

closed 2017-03-15 by fnoop