mavlink-router/mavlink-router #32: tcpendpoint doesn't retry on lost link

fnoop, 2017-03-16:
Using the new tcpendpoint when connecting as a client, if the remote/server drops connection (for example if the flight controller reboots or SITL restarts) then mavlink-router loses the connection forever.
Ideally, it should retry periodically to regain the connection.

lucasdemarchi, 2017-03-16:
A RetryTimeout value in the TcpEndpoint entry in the config file may make sense. Then we default it to 5s or so.

edersondisouza, 2017-03-28:
@fnoop Once #34 was merged, do you believe that this issue can be closed?

fnoop, 2017-04-01:
Confirmed fixed now, thanks :)

fnoop, 2017-04-01:
Oh, except when it does reconnect the mavlink-routerd process goes to 100% cpu

edersondisouza, 2017-04-03:
Almost there =D 
#57 should fix that!
Thanks for reporting!

edersondisouza, 2017-04-07:
@fnoop is it all working well now? Can we close this issue?

fnoop, 2017-04-07:
Yup confirmed fixed, thanks :)

closed 2017-04-07 by fnoop