mavlink-router/mavlink-router #33: Log error flooding

fnoop, 2017-03-16:
```
mavlinkrouter.sh[1342]: (endpoint.cpp:228) Different system_id message for endpoint 5: Current: 254 Read: 57
```
One per second

edersondisouza, 2017-03-17:
Flooding is a problem itself, but, this error itself needs to be treated: mavlink-router currently associates a single system_id with an endpoint, and clearly this assumption is not sound. That's why this warning is in place, after all.
@fnoop just for curiosity, do you know what are those system_id, `254` and `57`?

zehortigoza, 2017-03-17:
Probably this is happening because a corrupted Mavlink message was read. So it will have a random msg_id then this line:
`msg_entry = mavlink_get_msg_entry(msg_id);`
will return null and no CRC check will not run. So all other fields can have any garbage like the system_id field.

Well at least this happens at AeroFC & AeroCompute board

edersondisouza, 2017-03-28:
This also sounds bad - we accept that `mavlink_get_msg_entry(msg_id)` returns `NULL` for the case of a legitimate unknown id, but corrupt messages take all the fun.

fnoop, 2017-03-29:
@edersondisouza I don't know what those system_ids are, but the assumption that there is a single system_id to an endpoint is incorrect, I think - mavlink is a routed protocol.  You can have any number of components connected to a system that can/will route between other components on other connections.  You might want to know about this maybe in a debug mode, but certainly not in default run mode.

lucasdemarchi, 2017-03-29:
@fnoop the key word here is "component". Yes we support multiple components in an endpoint. But we used the constrain of one system. Otherwise it would be like if you have 2 drones sharing the same link and using the same endpoint which is a very weird situation. There could be other cases... E.g. a gimbal with different systemid?
But I really think there's a bug somewhere, either on our end or on the flight stack since I get this message too running just the flight stack and nothing else.

fnoop, 2017-03-29:
@lucasdemarchi Yes in an ideal world, however systemid/componentid is poorly understood/used and you often have a wide variety of them running on a single vehicle!

lucasdemarchi, 2017-03-29:
yep... eventually we will end up to deal with it. But I want to know if it's a misbehaving part using the wrong systemid or there's any bug hidden that we are not handling correctly.

lucasdemarchi, 2017-04-25:
Fixed on master

closed 2017-04-25 by lucasdemarchi