mavlink-router/mavlink-router #39: Allow some Endpoints to listen to messages based on a filter

edersondisouza, 2017-03-17:
Filtering would be made on routing function - it would allow an endpoint to even listen all messages, even those not meant for it.
A practical example would be LogEndpoints: a filter would avoid then receiving any messages not meant for logging.

Jaeyoung-Lim, 2018-08-14:
Is there any progress on this?

lucasdemarchi, 2018-08-14:
This is a feature idea we had a long time ago. It would be very welcome, but AFAIK no one is working on ti.

dk7xe, 2019-03-07:
Filtering would be really appreaciated on message id.
e.g. with the implementation of UTM_GLOBAL_POSITION message https://github.com/mavlink/mavlink/pull/1009 we have a message to report vehicle positions to air traffic management / UTM systems.
Currently all mavlink messages are forwarded to the UTM system which causes a lot of overhead.
Therefore filtering only needed messages in such a case would reduce the traffic a lot.

Jaeyoung-Lim, 2019-05-14:
@dk7xe This has been implemented on #167

closed 2019-05-14 by lucasdemarchi

dk7xe, 2019-05-23:
@Jaeyoung-Lim thank you!