mavlink-router/mavlink-router #49: Output to log file?

fnoop, 2017-03-29:
Currently all output goes to stdout, so if launched from systemd then it goes into the murky journal world.
It would be nice to output to log, at least as an option.

lucasdemarchi, 2017-03-29:
Can you just set the log location in your .service. not adding support for a file was done on purpose because of this.

fnoop, 2017-03-29:
Erm, how can you do that?  I did look through the depths of systemd but couldn't see a way to do that.  Everything I saw indicates that logging goes to journalctl.

lucasdemarchi, 2017-03-29:
I meant... not done on purpose because you have better ways than logging to a file, not that you are able to do that with systemd ;-). StandardOutput in systemd.exec man page gives you all the options. I usually use journal or tty or a combination thereof.

Why do you want to redirect to a file? Should we then do log rotation or overwrite what was there? Or a never-ending appending to that file? How to separate logs from last boot from the previous one? This is basically a solved problem with the journal and you can just do a "journalctl -u mavlink-routerd -b". And it controls rotation and how much space the logs can occupy, etc etc. And I can see the logs for all daemons in the system without fiddling with different locations for each of them.

fnoop, 2017-03-29:
Ah you got me all hopeful there!
OK that's fine, I'll just have to get used to journalctl :)

closed 2017-03-29 by fnoop