mavlink-router/mavlink-router #67: uartendpoint not working

fnoop, 2017-04-09:
[UartEndpoint serialin]
    Device=/dev/ttyS1
    Baud=115200

Proxy of mavlink not working as expected.  Source is definitely working, mavproxy with identical settings works as expected.

```
Apr 09 11:12:01 fnoop-joule mavlink-routerd[22045]: (endpoint.cpp:374) Open UART [4] /dev/ttyS1:115200 *
Apr 09 11:12:01 fnoop-joule mavlink-routerd[22045]: (endpoint.cpp:463) Open UDP [5] 0.0.0.0:14570 *
```

zehortigoza, 2017-04-10:
Please give more information, in our drone we use the uart as source and it works fine.

lucasdemarchi, 2017-04-10:
I just tested master branch and it seems to work fine with the following configuration file:

```
[General]
TcpServerPort=5760
ReportStats=false

[UartEndpoint uart]
Device = /dev/ttyS1
Baud = 460800

[UdpEndpoint wifi]
Mode = Normal
Address = 192.168.8.255

[UdpEndpoint ethernet-over-usb]
Mode = Normal
Address = 192.168.7.255
```

Are you sure you are using the right tty?

fnoop, 2017-04-10:
Yes, I'm using /dev/ttyS1 115200 baud on the joule.  Identical settings works on mavproxy, run back to back without reboot or any other changes.  Could you give me a clue to produce something useful to help debug?

lucasdemarchi, 2017-04-10:
Checking what exactly mavlink-routerd is doing would be my first debug:

```
strace -e trace=desc -e read=4 -p $(pidof mavlink-routerd)
```

In which 4 is the fd number you got from `Open UART [4]`

fnoop, 2017-04-14:
```
$ sudo strace -ff -p 5056
strace: Process 5056 attached
epoll_wait(3,
```
Not a lot..

Nothing else is using it:
```
$ lsof |grep ttyS1
mavlink-r 5056             mav    4u      CHR               4,65      0t0         88 /dev/ttyS1
```

fnoop, 2017-04-14:
```
/usr/bin/python /usr/local/bin/mavproxy.py --master /dev/ttyS1 --baud 115200  --out=udpin:0.0.0.0:14570 --out=udpin:0.0.0.0:14571 --out=udpin:0.0.0.0:14572 --out=udpin:0.0.0.0:14573 --out=udpin:0.0.0.0:14574 --out=tcpin:0.0.0.0:5780 --out=tcpin:0.0.0.0:5781 --out=tcpin:0.0.0.0:5782 --out=tcpin:0.0.0.0:5783 --out=tcpin:0.0.0.0:5784
```
This works fine on the same system without a reboot or any other reconfiguration.

lucasdemarchi, 2017-04-14:
Humn... And what's the config file for mavlink-router?

You were using mavlink-router before... Do you know a commit in which it works for you?

fnoop, 2017-04-22:
I've just tried latest master, same thing.  I can't remember which commit it worked, sorry.  I've run it under strace, and can see it open the serial port:
```
open("/dev/ttyS1", O_RDWR|O_NOCTTY|O_NONBLOCK|O_CLOEXEC) = 4
ioctl(4, TCGETS2, {B115200 -opost -isig -icanon -echo ...}) = 0
ioctl(4, TCSETS2, {0x1000 /* B??? */ -opost -isig -icanon -echo ...}) = 0
ioctl(4, TCFLSH, TCIOFLUSH)             = 0
```
It then seems to read some data in from serial:
```
read(4, "\376\32", 1120)                = 2
epoll_wait(3, [{EPOLLIN, {u32=7020560, u64=7020560}}], 8, -1) = 1
read(4, "\356\1\1\33\211R#`\0\0\0\0\27\0\334\377\26\374\0\0\340\377\7\0k\377\233\0?\2\207/"..., 1118) = 731
epoll_wait(3, [{EPOLLIN, {u32=7020560, u64=7020560}}], 8, -1) = 1
read(4, "\376\32\6\1\1\33\36\374&`\0\0\0\0\27\0\335\377\27\374\0\0\340\377\5\0j\377\233\0@\2"..., 1120) = 733
epoll_wait(3, [{EPOLLIN, {u32=7020560, u64=7020560}}], 8, -1) = 1
read(4, "\376\32\36\1\1\33\241\245*`\0\0\0\0\27\0\335\377\31\374\377\377\341\377\6\0i\377\233\0?\2"..., 1120) = 733
epoll_wait(3, [{EPOLLIN, {u32=7020560, u64=7020560}}], 8, -1) = 1
read(4, "\376\0326\1\1\33\22Q.`\0\0\0\0\25\0\335\377\31\374\0\0\341\377\6\0i\377\234\0@\2"..., 1120) = 733
epoll_wait(3, [{EPOLLIN, {u32=7020560, u64=7020560}}], 8, -1) = 1
read(4, "\376\tN\1\1\0\0\0\0\0\2\3Q\5\3\300%", 1120) = 17
epoll_wait(3, [{EPOLLIN, {u32=7020560, u64=7020560}}], 8, -1) = 1
```

At some point however it stops and sits in a timer/poll loop with nothing else coming in:
```
read(4, "\376\32\274\1\1\33i\17i`\0\0\0\0\26\0\336\377\25\374\377\377\340\377\6\0k\377\233\0?\2"..., 1120) = 733
epoll_wait(3, [{EPOLLIN, {u32=7020560, u64=7020560}}], 8, -1) = 1
read(4, "\376\32\324\1\1\33\225\275l`\0\0\0\0\27\0\335\377\26\374\0\0\340\377\5\0j\377\233\0?\2"..., 1120) = 733
epoll_wait(3, [{EPOLLIN, {u32=7077472, u64=7077472}}], 8, -1) = 1
read(11, "\1\0\0\0\0\0\0\0", 8)         = 8
epoll_wait(3, [{EPOLLIN, {u32=7077472, u64=7077472}}], 8, -1) = 1
read(11, "\1\0\0\0\0\0\0\0", 8)         = 8
epoll_wait(3, [{EPOLLIN, {u32=7077472, u64=7077472}}], 8, -1) = 1
read(11, "\1\0\0\0\0\0\0\0", 8)         = 8
epoll_wait(3, [{EPOLLIN, {u32=7077472, u64=7077472}}], 8, -1) = 1
read(11, "\1\0\0\0\0\0\0\0", 8)         = 8
epoll_wait(3, [{EPOLLIN, {u32=7077472, u64=7077472}}], 8, -1) = 1
read(11, "\1\0\0\0\0\0\0\0", 8)         = 8
epoll_wait(3, [{EPOLLIN, {u32=7077472, u64=7077472}}], 8, -1) = 1
read(11, "\1\0\0\0\0\0\0\0", 8)         = 8
```
In case it matters, ctsrts is set to off for this link (within ardupilot).  ttyS1 did have a getty running on it but I've stopped it and didn't make any different.  I can shut mavlink-router down and startup mavproxy with identical settings and it talks to ardupilot perfectly over the same serial link (even with getty running).

fnoop, 2017-04-22:
I notice mavproxy opens the serial port and turns off DTR/RTS:
```
open("/dev/ttyS1", O_RDWR|O_NOCTTY|O_NONBLOCK) = 3
ioctl(3, TCGETS, {B115200 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B115200 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B115200 -opost -isig -icanon -echo ...}) = 0
ioctl(3, SNDCTL_TMR_START or TCSETS, {B1200 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TCGETS, {B1200 -opost -isig -icanon -echo ...}) = 0
ioctl(3, TIOCMBIS, [TIOCM_DTR])         = 0
ioctl(3, TIOCMBIS, [TIOCM_RTS])         = 0
ioctl(3, TCFLSH, TCIFLUSH)              = 0
pipe([4, 6])                            = 0
pipe([7, 8])                            = 0
fcntl(4, F_SETFL, O_RDONLY|O_NONBLOCK)  = 0
fcntl(7, F_SETFL, O_RDONLY|O_NONBLOCK)  = 0
```

However as I write this mavlink-router has suddenly sprung into life and is working, and I can't reproduce the error.  So, sorry, I'll close this now and will reopen if I can reproduce it.

fnoop, 2017-04-22:
Ops, no it's reoccurred again after a reboot of both flight controller and companion computer.  I notice now it has high loss on both uart and udp endpoints:
```
Apr 22 10:40:42 fnoop-joule mavlink-routerd[3167]: Endpoint UART [4] sysid: 1 {
Apr 22 10:40:42 fnoop-joule mavlink-routerd[3167]:         Received messages {
Apr 22 10:40:42 fnoop-joule mavlink-routerd[3167]:                 CRC error: 2 0% 0KBytes
Apr 22 10:40:42 fnoop-joule mavlink-routerd[3167]:                 Sequence lost: 271 20%
Apr 22 10:40:42 fnoop-joule mavlink-routerd[3167]:                 Handled: 1077 35KBytes
Apr 22 10:40:42 fnoop-joule mavlink-routerd[3167]:                 Total: 1350
Apr 22 10:40:42 fnoop-joule mavlink-routerd[3167]:         }
Apr 22 10:40:42 fnoop-joule mavlink-routerd[3167]:         Transmitted messages {
Apr 22 10:40:42 fnoop-joule mavlink-routerd[3167]:                 Total: 88 1KBytes
Apr 22 10:40:42 fnoop-joule mavlink-routerd[3167]:         }
Apr 22 10:40:42 fnoop-joule mavlink-routerd[3167]: }
Apr 22 10:40:42 fnoop-joule mavlink-routerd[3167]: Endpoint UDP [5] sysid: 252 {
Apr 22 10:40:42 fnoop-joule mavlink-routerd[3167]:         Received messages {
Apr 22 10:40:42 fnoop-joule mavlink-routerd[3167]:                 CRC error: 0 0% 0KBytes
Apr 22 10:40:42 fnoop-joule mavlink-routerd[3167]:                 Sequence lost: 6606 98%
Apr 22 10:40:45 fnoop-joule mavlink-routerd[3167]:                 Handled: 88 1KBytes
Apr 22 10:40:45 fnoop-joule mavlink-routerd[3167]:                 Total: 6694
Apr 22 10:40:45 fnoop-joule mavlink-routerd[3167]:         }
Apr 22 10:40:45 fnoop-joule mavlink-routerd[3167]:         Transmitted messages {
Apr 22 10:40:45 fnoop-joule mavlink-routerd[3167]:                 Total: 438 14KBytes
Apr 22 10:40:45 fnoop-joule mavlink-routerd[3167]:         }
Apr 22 10:40:45 fnoop-joule mavlink-routerd[3167]: }
```

fnoop, 2017-04-22:
And if I restart the daemon, now it's working:
```
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]: Endpoint UART [4] sysid: 1 {
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         Received messages {
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 CRC error: 0 0% 0KBytes
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Sequence lost: 0 0%
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Handled: 3435 112KBytes
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Total: 3435
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         }
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         Transmitted messages {
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Total: 24 0KBytes
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         }
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]: }
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]: Endpoint UDP [5] sysid: 252 {
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         Received messages {
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 CRC error: 0 0% 0KBytes
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Sequence lost: 1524 98%
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Handled: 24 0KBytes
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Total: 1548
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         }
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         Transmitted messages {
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Total: 292 9KBytes
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         }
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]: }
```
Note no more errors on the serial link, but still a high 'Sequence lost' on the udp endpoint.

lucasdemarchi, 2017-04-22:
@fnoop can you connect mavproxy to one of these endpoints and look at the sequence lost number there?  I'm suspicious this number is wrong in mavlink-router... is it changing after some time or is it high in the beginning already?

Would be nice to check the following scenarios:

1)
* mavlink-router connected to the uart
* mavproxy connected to mavlink-router via udp
* check sequence lost on mavproxy

2)
* mavproxy connected to uart
* mavproxy connected to mavproxy via udp
* check sequence lost on mavproxy connected via udp

rachelruijiayang, 2017-04-23:
@fnoop What command did you run to read this?
```
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]: Endpoint UART [4] sysid: 1 {
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         Received messages {
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 CRC error: 0 0% 0KBytes
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Sequence lost: 0 0%
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Handled: 3435 112KBytes
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Total: 3435
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         }
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         Transmitted messages {
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Total: 24 0KBytes
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         }
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]: }
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]: Endpoint UDP [5] sysid: 252 {
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         Received messages {
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 CRC error: 0 0% 0KBytes
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Sequence lost: 1524 98%
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Handled: 24 0KBytes
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Total: 1548
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         }
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         Transmitted messages {
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:                 Total: 292 9KBytes
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]:         }
Apr 22 10:44:43 fnoop-joule mavlink-routerd[3238]: }
```

fnoop, 2017-04-23:
OK well I tried turning off RTS in the uartendpoint code, and it doesn't make any difference.
I note it seems to now work OK if I restart mavlink-router, it's the first run, oddly, that doesn't work.  Although in the past I'm sure multiple restarts didn't make it work.

fnoop, 2017-04-23:
@lucasdemarchi OK so restarting mavlink-router it sometimes works, sometimes doesn't - ie. it's intermittent.  If mavlink-router isn't working, it takes a while to restart it (using systemd), if it's working it restarts immediately.
When it's not working, mavproxy can't connect to it at all, when it is working mavproxy can connect to it just fine and the link is 100% ok (ie. 0% loss), so it's all or nothing which is the same thing I see when trying to connect a GCS to mavlink-router.  It either works perfectly or not at all.

lucasdemarchi, 2017-04-24:
@fnoop I think there's something different with your setup that I'm not able to reproduced. Besides this working on Aero RTF, I tried to test this with the following setup:

* mavlink-router from master branch
* pixhawk with ArduPilot connected via TELEM1 port to a usbserial
* usbserial connected to my host computer

mavlink-routerd is started with:

```bash
./mavlink-routerd -r -e 127.0.0.1:5000 /dev/ttyUSB0:115200
```

I can succesfully connect mavproxy to 127.0.0.1:5000 or the examples/heartbeat-print.py script

I set the params on ArduPilot to match your setup: 
SERIAL1_BAUD     115.000000
BRD_SER1_RTSCTS  0.000000

I did try setting the uart settings to something else before starting mavlink-router

Can you post the output of `stty -F /dev/ttyS1` before and after starting mavlink-router/mavproxy?  Do you have the flowcontrol pins physically connected?

fnoop, 2017-04-24:
After boot, before running anything on ttyS1:
```
[dev] [mav@fnoop-joule ~]$ stty -F /dev/ttyS1
speed 9600 baud; line = 0;
-brkint -imaxbel
```
After starting mavlink-router for the first time (and it not responding to proxy requests):
```
[dev] [mav@fnoop-joule ~]$stty -F /dev/ttyS1c
speed 115200 baud; line = 0;
min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo
```
And after restarting a couple of times until it works:
```
[dev] [mav@fnoop-joule ~]$stty -F /dev/ttyS1c
speed 115200 baud; line = 0;
min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo
```

fnoop, 2017-04-24:
This is after starting mavproxy (working, 100% reliable):
```
[dev] [mav@fnoop-joule ~]$ tty -F /dev/ttyS1
speed 115200 baud; line = 0;
min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo -echoe -echok -echoctl -echoke
```

fnoop, 2017-04-24:
So connecting a GCS to working mavlink-router session gets horizon/data from the flight controller proxied as expected.  However, disconnecting and reconnecting the GCS a couple of times then stalls.  Here is the point at which the link dies and doesn't come back:
```
epoll_wait(3, [{EPOLLIN, {u32=16719888, u64=16719888}}], 8, -1) = 1
read(4, "\376\34\5\1\1\36\254\330=\0U\306\230=\211$\333\274\220\354\233>\\\177\204\272b\221\317\274\3715"..., 1120) = 144
sendto(5, "\376\34\5\1\1\36\254\330=\0U\306\230=\211$\333\274\220\354\233>\\\177\204\272b\221\317\274\3715"..., 36, 0, {sa_family=AF_INET, sin_port=htons(55580), sin_addr=inet_addr("192.168.1.137")}, 16) = 36
sendto(5, "\376\30\6\1\1\262~\314\271=\243u\354\274zQ\255>\0\0\0\0\0\0\0\0\0\0\0\0\215U", 32, 0, {sa_family=AF_INET, sin_port=htons(55580), sin_addr=inet_addr("192.168.1.137")}, 16) = 32
sendto(5, "\376(\7\1\1\266U\306\230=\211$\333\274\220\354\233>\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 48, 0, {sa_family=AF_INET, sin_port=htons(55580), sin_addr=inet_addr("192.168.1.137")}, 16) = 48
sendto(5, "\376\24\10\1\1J\0\0\0\0\307\4K<=\n7@\0\0\0\0\21\0\0\0A=", 28, 0, {sa_family=AF_INET, sin_port=htons(55580), sin_addr=inet_addr("192.168.1.137")}, 16) = 28
epoll_wait(3, [{EPOLLIN, {u32=16719888, u64=16719888}}], 8, -1) = 1
read(4, "\376\34\t\1\1\36\17\331=\0\270\307\230=%}\333\274\216\357\233> \304\334\271BD\320\274\221i"..., 1120) = 144
sendto(5, "\376\34\t\1\1\36\17\331=\0\270\307\230=%}\333\274\216\357\233> \304\334\271BD\320\274\221i"..., 36, 0, {sa_family=AF_INET, sin_port=htons(55580), sin_addr=inet_addr("192.168.1.137")}, 16) = 36
sendto(5, "\376\30\n\1\1\262\235\351\271=\16\355\353\274,X\255>\0\0\0\0\0\0\0\0\0\0\0\0\35\352", 32, 0, {sa_family=AF_INET, sin_port=htons(55580), sin_addr=inet_addr("192.168.1.137")}, 16) = 32
sendto(5, "\376(\v\1\1\266\270\307\230=%}\333\274\216\357\233>\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 48, 0, {sa_family=AF_INET, sin_port=htons(55580), sin_addr=inet_addr("192.168.1.137")}, 16) = 48
sendto(5, "\376\24\f\1\1J\0\0\0\0J4?<=\n7@\0\0\0\0\21\0\0\0Q\r", 28, 0, {sa_family=AF_INET, sin_port=htons(55580), sin_addr=inet_addr("192.168.1.137")}, 16) = 28
epoll_wait(3, [{EPOLLIN, {u32=16719888, u64=16719888}}], 8, -1) = 1
read(4, "\376\34\r\1\1\36t\331=\0\16\303\230=\351\323\333\274T\357\233>\32H\206\272\237\330\327\274\207#"..., 1120) = 144
sendto(5, "\376\34\r\1\1\36t\331=\0\16\303\230=\351\323\333\274T\357\233>\32H\206\272\237\330\327\274\207#"..., 36, 0, {sa_family=AF_INET, sin_port=htons(55580), sin_addr=inet_addr("192.168.1.137")}, 16) = 36
sendto(5, "\376\30\16\1\1\262\33\331\271={\217\353\274\207V\255>\0\0\0\0\0\0\0\0\0\0\0\0\\\r", 32, 0, {sa_family=AF_INET, sin_port=htons(55580), sin_addr=inet_addr("192.168.1.137")}, 16) = 32
sendto(5, "\376(\17\1\1\266\16\303\230=\351\323\333\274T\357\233>\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 48, 0, {sa_family=AF_INET, sin_port=htons(55580), sin_addr=inet_addr("192.168.1.137")}, 16) = 48
sendto(5, "\376\24\20\1\1J\0\0\0\0,\2302<=\n7@\0\0\0\0\21\0\0\0\342\f", 28, 0, {sa_family=AF_INET, sin_port=htons(55580), sin_addr=inet_addr("192.168.1.137")}, 16) = 28
epoll_wait(3, [{EPOLLIN, {u32=16776800, u64=16776800}}], 8, -1) = 1
read(11, "\1\0\0\0\0\0\0\0", 8)         = 8
epoll_wait(3, [{EPOLLIN, {u32=16776800, u64=16776800}}], 8, -1) = 1
read(11, "\1\0\0\0\0\0\0\0", 8)         = 8
epoll_wait(3, [{EPOLLIN, {u32=16719392, u64=16719392}}], 8, -1) = 1
recvfrom(5, "HELLO", 1120, 0, {sa_family=AF_INET, sin_port=htons(51006), sin_addr=inet_addr("192.168.1.137")}, [16]) = 5
epoll_wait(3, [{EPOLLIN, {u32=16719392, u64=16719392}}], 8, -1) = 1
recvfrom(5, "\376\t`\374\1\0\0\0\0\0\6\10\300\4\3k\352", 1120, 0, {sa_family=AF_INET, sin_port=htons(51006), sin_addr=inet_addr("192.168.1.137")}, [16]) = 17
write(4, "\376\t`\374\1\0\0\0\0\0\6\10\300\4\3k\352", 17) = 17
epoll_wait(3, [{EPOLLIN, {u32=16719392, u64=16719392}}], 8, -1) = 1
recvfrom(5, "\376\6a\374\1B\2\0\1\0\2\1|\327", 1120, 0, {sa_family=AF_INET, sin_port=htons(51006), sin_addr=inet_addr("192.168.1.137")}, [16]) = 14
write(4, "\376\6a\374\1B\2\0\1\0\2\1|\327", 14) = 14
epoll_wait(3, [{EPOLLIN, {u32=16719392, u64=16719392}}], 8, -1) = 1
recvfrom(5, "\376\6b\374\1B\3\0\1\0\6\1{\325", 1120, 0, {sa_family=AF_INET, sin_port=htons(51006), sin_addr=inet_addr("192.168.1.137")}, [16]) = 14
write(4, "\376\6b\374\1B\3\0\1\0\6\1{\325", 14) = 14
epoll_wait(3, [{EPOLLIN, {u32=16719392, u64=16719392}}], 8, -1) = 1
recvfrom(5, "\376\6c\374\1B\n\0\1\0\n\1pd", 1120, 0, {sa_family=AF_INET, sin_port=htons(51006), sin_addr=inet_addr("192.168.1.137")}, [16]) = 14
write(4, "\376\6c\374\1B\n\0\1\0\n\1pd", 14) = 14
epoll_wait(3, [{EPOLLIN, {u32=16719392, u64=16719392}}], 8, -1) = 1
recvfrom(5, "\376\6c\374\1B\n\0\1\0\n\1pd", 1120, 0, {sa_family=AF_INET, sin_port=htons(51006), sin_addr=inet_addr("192.168.1.137")}, [16]) = 14
write(4, "\376\6c\374\1B\n\0\1\0\n\1pd", 14) = 14
epoll_wait(3, [{EPOLLIN, {u32=16719392, u64=16719392}}], 8, -1) = 1
recvfrom(5, "\376\6d\374\1B\n\0\1\0\v\1J\236", 1120, 0, {sa_family=AF_INET, sin_port=htons(51006), sin_addr=inet_addr("192.168.1.137")}, [16]) = 14
write(4, "\376\6d\374\1B\n\0\1\0\v\1J\236", 14) = 14
epoll_wait(3, [{EPOLLIN, {u32=16719392, u64=16719392}}], 8, -1) = 1
recvfrom(5, "\376\6d\374\1B\n\0\1\0\v\1J\236", 1120, 0, {sa_family=AF_INET, sin_port=htons(51006), sin_addr=inet_addr("192.168.1.137")}, [16]) = 14
write(4, "\376\6d\374\1B\n\0\1\0\v\1J\236", 14) = 14
epoll_wait(3, [{EPOLLIN, {u32=16719392, u64=16719392}}], 8, -1) = 1
[... cut ...]
epoll_wait(3, [{EPOLLIN, {u32=16719392, u64=16719392}}], 8, -1) = 1
recvfrom(5, "\376\6f\374\1B\2\0\1\0\1\1\376\230", 1120, 0, {sa_family=AF_INET, sin_port=htons(51006), sin_addr=inet_addr("192.168.1.137")}, [16]) = 14
write(4, "\376\6f\374\1B\2\0\1\0\1\1\376\230", 14) = 14
epoll_wait(3, [{EPOLLIN, {u32=16719392, u64=16719392}}], 8, -1) = 1
recvfrom(5, "\376\6g\374\1B\2\0\1\0\3\1\327x", 1120, 0, {sa_family=AF_INET, sin_port=htons(51006), sin_addr=inet_addr("192.168.1.137")}, [16]) = 14
write(4, "\376\6g\374\1B\2\0\1\0\3\1\327x", 14) = 14
```

lucasdemarchi, 2017-04-27:
@fnoop  can you test if #77 helps with your issue?  I couldn't test it on a Joule yet, but I hope this fixes it.

fnoop, 2017-04-27:
@lucasdemarchi  I'm afraid not:
```
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]: Endpoint UART [9] sysid: 1 {
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:         Received messages {
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:                 CRC error: 46 4% 1KBytes
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:                 Sequence lost: 628 66%
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:                 Handled: 272 9KBytes
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:                 Total: 946
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:         }
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:         Transmitted messages {
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:                 Total: 12 0KBytes
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:         }
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]: }
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]: Endpoint UDP [4] sysid: 252 {
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:         Received messages {
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:                 CRC error: 0 0% 0KBytes
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:                 Sequence lost: 762 98%
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:                 Handled: 12 0KBytes
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:                 Total: 774
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:         }
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:         Transmitted messages {
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:                 Total: 0 0KBytes
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]:         }
Apr 27 10:02:34 maverick-joule mavlink-routerd[21615]: }
```
Stopping mavlink-router and starting mavproxy without doing anything else, mavproxy works with 0% loss on the link.

I note from my strace of mavproxy above that it actually seems to turn ON flow control?
```
ioctl(3, TIOCMBIS, [TIOCM_DTR])         = 0
ioctl(3, TIOCMBIS, [TIOCM_RTS])         = 0
```

lucasdemarchi, 2017-04-28:
Dunno how that could help, but I added it nonetheless to that PR.

lucasdemarchi, 2017-05-09:
One thing that could be happening is the underlying device dropping.  This should be better handled by mavlink-router with a big warning about this and probably try to reopen the fd like is done for TCP reconnection.

Anyway, the issue for *this* ticket seems to be fixed

closed 2017-05-09 by lucasdemarchi