mavlink-router/mavlink-router #70: 20s drops in logging

ChristophTobler, 2017-04-13:
When logging with the mavlink-router, it can happen that I have drops of 20 seconds. see e.g. lpos.z in http://logs.px4.io/plot_app?log=bb8372b4-60bc-4c7c-94a4-b1cdddf0efb0

lucasdemarchi, 2017-04-13:
@ChristophTobler where are you getting the logs from? From mavlink-router itself or on ground? are you using master branch from this repo?

On the logs you sent I actually see a max interval of 1.6s between any 2 imu samples. So we are certainly receiving data. Not that that 1.6s is a good thing, but there's something else going on, it's certainly not "20sec without receiving any sample".  Would be good to have the perf counters and know if the logger dropped on the stm32 side.

@edersondisouza can you review the logging recording for PX4 and check it's as it should be?

ChristophTobler, 2017-04-13:
@lucasdemarchi I use `mavlink-router ... --log` to log. I used master that is maybe 2-3 weeks old. I didn't say 20s without any samples. Please check local position Z for example there are drops as big as 20s (I think somewhere about 80s to 100s). Generally data is not logging regularly.

lucasdemarchi, 2017-05-09:
This doesn't seem related to what mavlink-router is doing but rather how px4 is prioritizing packets on a  link with losses.  If you stop mavlink-router and use only `mavlink_ulog_streaming.py` on the UART, is it any better?

ChristophTobler, 2017-05-10:
Ok, I guess the bandwidth is just too low to have full logging rate. I don't think it's the mavlink-router but the low rate of the port. Having a stable higher rate is crucial for the logging.

closed 2017-05-10 by lucasdemarchi

ChristophTobler, 2017-05-11:
Just for other people stumbling upon this: In the meantime, the general rate/interval can be lowered/increased [here](https://github.com/PX4/Firmware/blob/08660251499ceaaf613706232d99138622dd6197/src/modules/logger/logger.cpp#L264) to have more constant logging. Or individual topics [here](https://github.com/PX4/Firmware/blob/08660251499ceaaf613706232d99138622dd6197/src/modules/logger/logger.cpp#L565-L609)