mavlink-router/mavlink-router #74: TCP port 5760 not receiving routed messages

broody, 2017-04-24:
For some reason in latest I'm not receiving any mavlink messages on the local TCP port anymore. Connection establishes successfully, but nothing is received afterwards.

broody, 2017-04-25:
Ah sorry, I am getting the messages, except my mavlink_parse_char() function is no longer returns valid message. I'll debug further on my side.

closed 2017-04-25 by broody

lucasdemarchi, 2017-04-25:
@broody do you know the last version in which it was working?

broody, 2017-04-25:
@lucasdemarchi It was working on v01.00.04, let me double check this.

lucasdemarchi, 2017-04-25:
that version of BSP was not actually using mavlink-router

lucasdemarchi, 2017-04-25:
Quick test here on my computer with a pixhawk connected to a usbserial:

Change `examples/heartbeat-print.py` to use TCP:

```
mav = mavutil.mavlink_connection('tcp:' + sys.argv[1])
```

Start mavlink-router:

```
./mavlink-routerd -e 127.0.0.1:14550 /dev/ttyUSB0:921600
```

running the example I get:
```
./examples/heartbeat-print.py 127.0.0.1:5760
HEARTBEAT {type : 2, autopilot : 3, base_mode : 81, custom_mode : 0, system_status : 3, mavlink_version : 3}
HEARTBEAT {type : 2, autopilot : 3, base_mode : 81, custom_mode : 0, system_status : 3, mavlink_version : 3}
HEARTBEAT {type : 2, autopilot : 3, base_mode : 81, custom_mode : 0, system_status : 3, mavlink_version : 3}
```

Also works with a test config file:
```
$ cat /tmp/test.conf 
[General]
TcpServerPort=5760
ReportStats=false

[UartEndpoint flightstack]
Device = /dev/ttyUSB0
Baud = 921600

./examples/heartbeat-print.py 127.0.0.1:5760
HEARTBEAT {type : 2, autopilot : 3, base_mode : 81, custom_mode : 0, system_status : 3, mavlink_version : 3}
HEARTBEAT {type : 2, autopilot : 3, base_mode : 81, custom_mode : 0, system_status : 3, mavlink_version : 3}
HEARTBEAT {type : 2, autopilot : 3, base_mode : 81, custom_mode : 0, system_status : 3, mavlink_version : 3}

```

broody, 2017-04-25:
I just flashed v01.00.04 and remember now, it includes mavlink-routerd but it's the version without the local TCP port or conf file enabled. I had to pulled the latest mavlink-router (I think this was around March 22nd) and compile to enable local TCP port.   

Thanks for the test examples, I'll give these a try.

lucasdemarchi, 2017-04-25:
@broody you may find #75 useful for debugging

broody, 2017-04-27:
Ah it's working now. It was because I was using the v1.0 protocol. Looks like I need to use v2.0.