mavlink-router/mavlink-router #81: Handle HUP on UART

lucasdemarchi, 2017-05-09:
The tty we are using can generate a HUP (EPOLLIN | EPOLLERR | EPOLLHUP) in which case we should a) stop using that fd and b) probably try reopening it just like we try to reconnect to TCP.

@fnoop I think this is the remaining issue for you... One way to verify that is to search for "serial" in /proc/interrupts. Once the device goes away you will notice: 
a) you will no longer see "serial" on the correspondent irq and
b) strace -p $(pidof mavlink-routerd) will should epoll_wait returning `EPOLLIN | EPOLLERR | EPOLLHUP`

I think there's something bad going on on your kernel/device, but we should deal with it nonetheless

lucasdemarchi, 2017-05-09:
btw... the only way I've got to reproduce is by (accidentally) having 2 different process (mavlink-routerd and getty) fighting for the UART.

fnoop, 2017-05-11:
@lucasdemarchi  This makes a lot of sense, thanks.  I'll try and test it as soon as I can.