mavlink-router/mavlink-router #88: Move configuration parse to each class

lucasdemarchi, 2017-05-10:
The point of heaving ConfFile::OptionsTable was exactly to allow each class to have its own section with options.

We should move things out of `parse_confs()` into an specific class. Example:

```c++
    static const ConfFile::OptionsTable option_table_uart[] = {                                                        
        {"baud",    false,  ConfFile::parse_ul,         OPTIONS_TABLE_STRUCT_FIELD(option_uart, baud)},                
        {"device",  true,   ConfFile::parse_str_dup,    OPTIONS_TABLE_STRUCT_FIELD(option_uart, device)},              
    };      
```

This should be in UartEndpoint

... and so on.

jbeyerstedt, 2021-12-09:
This sounds like a good idea at first, but when thinking about an implementation, it didn't look so good to me any more. Since we have two sources of endpoint configuration (config file and CLI parameters), the endpoint would need to take care of all those variants.

I would rather like to have _one_ interface to give the endpoint it's configuration, which is independent from the actual source of configuration. So assembling the "merged" configuration is the responsibility of one part of the software and "executing" the config is the sole purpose of the rest. This split will also make testing easier.

The options tables themselves could be moved to the respective endpoints, though. Because in general, you're right, that the actual configuration parameters are the responsibility of the endpoint classes.

Because I started to dig though the setup code with the intention to also fix this issue. But with [the current state]https://github.com/jbeyerstedt/mavlink-router/tree/feature/configparsing-rework, I thinks it's quite usable again (PR will follow as soon as my cleanup PR is merged).

lucasdemarchi, 2021-12-09:
> The options tables themselves could be moved to the respective endpoints, though. Because in general, you're right, that the actual configuration parameters are the responsibility of the endpoint classes.

this is about moving the options tables ^. Idea is that the generic code matches a specific section with a specific class, gets the table from that specific class and does the assembly with command line.

jbeyerstedt, 2021-12-09:
Thanks, this makes more sense. I already have the log endpoint in a separate table as well, so this should be a quick addition to my upcoming PR.

commit 46ce5094 (2021-12-15): Config: Move option tables to endpoint classes (fixes #88)
closed 2021-12-15 by lucasdemarchi