mavlink-router/mavlink-router #90: WARN  [mavlink] ignoring CMD with same SYS/COMP (1/1) ID

svpcom, 2017-05-15:
PX4 v1.5 (stable) produces following messages in infinite loop:
```
WARN  [mavlink] ignoring CMD with same SYS/COMP (1/1) ID
WARN  [mavlink] ignoring CMD with same SYS/COMP (1/1) ID
WARN  [mavlink] ignoring CMD with same SYS/COMP (1/1) ID
WARN  [mavlink] ignoring CMD with same SYS/COMP (1/1) ID
WARN  [mavlink] ignoring CMD with same SYS/COMP (1/1) ID
WARN  [mavlink] ignoring CMD with same SYS/COMP (1/1) ID
```

svpcom, 2017-05-15:
main.conf:
``` .ini
[General]
TcpServerPort=0
ReportStats=false
MavlinkDialect=ardupilotmega

[UartEndpoint px4]
Device = /dev/ttyS1
Baud = 921600

[UdpEndpoint wifibroadcast]
Mode = Normal
Address = 127.0.0.1
Port = 14550

[UdpEndpoint qgc]
Mode = Normal
Address = 10.4.254.253
Port = 14550
```

lucasdemarchi, 2017-05-15:
What HW are you working on? Is this on mavlink-router from master branch?  My first suspect would be the logging system with wrong sysid/compid, but you don't have that enabled.... So, what do you have running on the wifibroadcast endpoint?

zehortigoza, 2017-05-16:
This is some message that PX4 send periodically with the wrong target system id.
Like this one: https://github.com/PX4/Firmware/pull/6896/commits/519ca279342d173c41b1eac6b057de06cc772476

svpcom, 2017-05-16:
HW is a odroid-c0, connected to pixhawk as companion at 921600 bps. mavlink-router
 was built from master branch, px4 is latest stable (not master). Wifibroadcast is a unidirectional transport - it only receives mavlink packets, but never send data back (https://github.com/svpcom/wifibroadcast)

lucasdemarchi, 2017-05-18:
@svpcom this seems to be a problem in PX4 not mavlink-router as @zehortigoza said.  The only message we send ourselves to the flight stack is to start the log streaming (if you have that enabled on the configuration file).

Btw, nice to see wifibroadcast alive again. I hope to be able to test it later.

closed 2017-05-18 by lucasdemarchi