mavlink-router/mavlink-router #95: 'Normal' udp endpoint doesn't work

fnoop, 2017-06-09:
So tcp endpoint works fine (I can connect client to defined tcp port and it gets routed mavlink correctly), and udp client works (I can connect client as udp client to 'eavesdropping' client ports), but a Normal udp endpoint doesn't seem to work at all.  No packets shown flowing through tcpdump, client (eg. dronekit) doesn't get any connection.

lucasdemarchi, 2017-06-09:
Do you have a correct sysid/compid (i.e. different from the flight stack)? Does it send a heartbeat or at least any message so mavlink-router knows its sysid/compid?

#92 should be fixing the compid handling.

lucasdemarchi, 2017-06-09:
Also, could you share the snippet of dronekit code you are using that reproduces the problem?

fnoop, 2017-06-09:
Code is here:
https://github.com/fnoop/vision_landing/blob/master/vision_landing#L205

Relevant connect string is 'udp:127.0.0.1:14570'

No, not changing the cmpid.  I assume dronekit takes care of heartbeat etc?

lucasdemarchi, 2017-06-29:
@fnoop it's working fine here. "Eavesdropping" is like "ignore every routing based on sysid/compid" and let me inspect everything.

"Normal" is for properly behaved components in which it allows the other side to learn it is there. In other words, it has to send at least one message with its sysid/compid for mavlink-router to learn that that endpoint has a "node" with that "address".

This commits details the routing:
https://github.com/01org/mavlink-router/commit/c714bb7c92c0c277e3963170abfbaaeff11420ac

We need to move this into proper documentation. Also notice that  before that commit we would route based on sysid only, ignoring compid (this is because compid is very little used and we had problems with it before).

I don't think dronekit will send any heartbeat for you, but it's not my area of expertise.

fnoop, 2017-06-30:
I'll look further into this, but as it stands, mavlink-router doesn't seem to work with dronekit.  I would have thought it would be more pragmatic to make sure it works with dronekit out of the box.

lucasdemarchi, 2017-06-30:
It actually does. I know people using it. The thing is that you can't receive targeted messages if you don't send any because then mavlink-router won't know about it.

Ederson, I remember you had a sample with dronekit... Can you add it to our examples?

fnoop, 2017-06-30:
My dronekit script fails on connection, even with a specified sysid (rather than the default 255).  If I stop mavlink-router and fire up mavproxy, with identical configuration, I can connect fine.

lucasdemarchi, 2017-06-30:
Let's reopen the issue then until we can give an example that works.

fnoop, 2018-02-27:
Ok this is a little embarrassing, but might be worth adding a warning about.  The config file was being generated with <n> udp endpoints and <n> udpin endpoints, with the endpoint name generated as well by an incrementer (eg. udpendpoint 0, udpendpoint 1 etc).  But the endpoint name wasn't being changed between the udp endpoints udpin endpoints, so I had:

```
[UdpEndpoint 0]
    Mode = normal
    Address = 127.0.0.1
    Port = 14580
[UdpEndpoint 1]
    Mode = normal
    Address = 127.0.0.1
    Port = 14581
[UdpEndpoint 0]
    Mode = eavesdropping
    Address = 0.0.0.0
    Port = 14583
[UdpEndpoint 1]
    Mode = eavesdropping
    Address = 0.0.0.0
    Port = 14584
```

It seems the first 'UdpEndpoint 0' was being overwritten by the second 'UdpEndpoint 0' so the first entry wasn't working.  This is of course stupid on my part, but at the same time it's taken me 6 months to find the problem, a warning in the log of a duplicate endpoint would be very useful to help users avoid this.

closed 2018-02-27 by fnoop