mavlink-router/mavlink-router #96: Log files all zero length

fnoop, 2017-06-09:
Using this config:
```
[General]
#Mavlink-router serves on this TCP port
TcpServerPort=5770
ReportStats=false
#Which mavlink dialect is being used. Can be `common`[default] or `ardupilotmega`
MavlinkDialect=auto
# error, warning, info or debug
DebugLogLevel=info
# Log - Path to directory where to store flightstack log.
Log=/srv/maverick/data/mavlink/fc

[UartEndpoint serialin]
    Device=/dev/ttyS1
    Baud=115200,115200,57600

[UdpEndpoint 0]
    Mode = Normal
    Address = 127.0.0.1
    Port = 14570
[UdpEndpoint 1]
    Mode = Normal
    Address = 127.0.0.1
    Port = 14571
[UdpEndpoint 2]
    Mode = Normal
    Address = 127.0.0.1
    Port = 14572

[UdpEndpoint 0]
    Mode = Eavesdropping
    Address = 0.0.0.0
    Port = 14573
[UdpEndpoint 1]
    Mode = Eavesdropping
    Address = 0.0.0.0
    Port = 14574
[UdpEndpoint 2]
    Mode = Eavesdropping
    Address = 0.0.0.0
    Port = 14575
```

After several flights, all log files are zero length:
```
[dev] [mav@maverick-joule ~/data/mavlink/fc]$ ls -l
total 0
-rw-r--r-- 1 mav mav 0 Jun  9 09:08 00000-2017-06-09_09-08-15.bin
-rw-r--r-- 1 mav mav 0 Jun  9 11:27 00001-2017-06-09_11-27-18.bin
-rw-r--r-- 1 mav mav 0 Jun  9 11:28 00002-2017-06-09_11-28-14.bin
-rw-r--r-- 1 mav mav 0 Jun  9 14:33 00003-2017-06-09_14-33-24.bin
-rw-r--r-- 1 mav mav 0 Jun  9 14:45 00004-2017-06-09_14-45-59.bin
-rw-r--r-- 1 mav mav 0 Jun  9 14:58 00005-2017-06-09_14-58-24.bin
-rw-r--r-- 1 mav mav 0 Jun  9 15:02 00006-2017-06-09_15-02-16.bin
-rw-r--r-- 1 mav mav 0 Jun  9 15:07 00007-2017-06-09_15-07-27.bin
```

Mavlink-router is routing mavlink, these flights were all using mavlink control script connected to mavlink-router.

lucasdemarchi, 2017-06-09:
Did you enable logging on the flight stack to be sent via mavlink? On ArduPilot that's [LOG_BACKEND_TYPE](http://ardupilot.org/copter/docs/parameters.html#log-backend-type-dataflash-backend-storage-type)

fnoop, 2017-06-09:
Ah, no :) Probably a good thing to document.
I'm guessing standard 115200 baud isn't going to cope with mavlink logging, is there a recommended minimum?

fnoop, 2017-06-09:
Confirmed now getting data in bin files when changed LOG_BACKEND_TYPE to 3.  Have also increased baud to 921600.  Will close when confirmed data is writing correctly.

lucasdemarchi, 2017-06-22:
@fnoop we use 460800, but if you can increase it to 921600 it's preferred and what others are using (example, 3DR Solo). Check for crc errors, if you don't have them, you are good.

lucasdemarchi, 2017-06-22:
> Confirmed now getting data in bin files when changed LOG_BACKEND_TYPE to 3. Have also increased baud to 921600. Will close when confirmed data is writing correctly.

ping

fnoop, 2017-06-22:
Sorry, I'm getting data through into the log files fine!  Using 921600, no problems.  Confirmed that the data is good and parses fine.

closed 2017-06-22 by fnoop

laurenweinstein1, 2020-05-04:
I'm trying to assist getting logging working through mavlink-router on a PX4 stack. There is no LOG_BACKEND_TYPE parameter. There is the bitmask for what to log, with a variety of items included. Mavlink-router is working fine routing packets to/from USB on the device to/from ground control stations like QGC (e.g., via WiFi or telemetry radios), which can connect fine. Logging is enabled on mavlink-router via -l, the specified directory exists, and empty .bin files are being created by mavlink-router. No data is being logged however, the files remain zero length. Any ideas? Thanks.