mavlink-router/mavlink-router #97: Buffer overflow while logging

nkhoit, 2017-06-13:
Trying to log results in a buffer overflow
mavlink-routerd -l . /dev/ttyTHS2:921600
Open UART [4] /dev/ttyTHS2 *
UART [4] speed = 921600
Open TCP [5] 0.0.0.0:5760 *
Logging target system_id=1 on 02021-2017-06-12_23-58-43.ulg
*** buffer overflow detected ***: mavlink-routerd terminated
Aborted

I'm running it on an Nvidia TX1 with Ubuntu 16.04.2 LTS (arm64) 
g++ --version
g++ (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609

edersondisouza, 2017-06-13:
Hi @nkhoit,

Do you have a core dump file? Or could you please run `mavlink-routerd` on valgrind or gdb? So we could get more information on this issue

Thanks

nkhoit, 2017-06-13:
Valgrind:
valgrind ./mavlink-router/mavlink-routerd -l . /dev/ttyTHS2:921600
==20435== Memcheck, a memory error detector
==20435== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==20435== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
==20435== Command: ./mavlink-router/mavlink-routerd -l . /dev/ttyTHS2:921600
==20435==
Open UART [4] /dev/ttyTHS2 *
UART [4] speed = 921600
Open TCP [5] 0.0.0.0:5760 *
Logging target system_id=1 on 00002-2017-06-13_20-50-29.ulg
==20435== Syscall param epoll_pwait(sigmask) points to unaddressable byte(s)
==20435==    at 0x4B10240: epoll_pwait (epoll_pwait.c:42)
==20435==    by 0x40E59F: Mainloop::loop() (mainloop.cpp:292)
==20435==    by 0x40296B: main (main.cpp:805)
==20435==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==20435==
**20435** *** memcpy_chk: buffer overflow detected ***: program terminated
==20435==    at 0x48479CC: ??? (in /usr/lib/valgrind/vgpreload_memcheck-arm64-linux.so)
==20435==
==20435== HEAP SUMMARY:
==20435==     in use at exit: 95,055 bytes in 17 blocks
==20435==   total heap usage: 34 allocs, 17 frees, 132,896 bytes allocated
==20435==
==20435== LEAK SUMMARY:
==20435==    definitely lost: 0 bytes in 0 blocks
==20435==    indirectly lost: 0 bytes in 0 blocks
==20435==      possibly lost: 0 bytes in 0 blocks
==20435==    still reachable: 95,055 bytes in 17 blocks
==20435==         suppressed: 0 bytes in 0 blocks
==20435== Rerun with --leak-check=full to see details of leaked memory
==20435==
==20435== For counts of detected and suppressed errors, rerun with: -v
==20435== ERROR SUMMARY: 400 errors from 1 contexts (suppressed: 0 from 0)

lucasdemarchi, 2017-06-30:
Can you try again with master branch. This is the most common case we test so it would be surprising if it's not working for you

nkhoit, 2017-06-30:
I tried again on master and I get the same problem on my Nvidia TX1.
I decided to try on my x86-64 Ubuntu 16.04 desktop and got a buffer overflow as well but a different valgrind output.
valgrind ./mavlink-routerd -l ~ /dev/ttyACM0
==27205== Memcheck, a memory error detector
==27205== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==27205== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
==27205== Command: ./mavlink-routerd -l /home/nkhoit /dev/ttyACM0
==27205== 
==27205== Invalid read of size 1
==27205==    at 0x40D84A: strlist_to_ul (main.cpp:268)
==27205==    by 0x40D84A: add_uart_endpoint(char const*, unsigned long, char const*, char const*) (main.cpp:327)
==27205==    by 0x4024F2: parse_argv (main.cpp:481)
==27205==    by 0x4024F2: main (main.cpp:790)
==27205==  Address 0x5ab6e7d is 0 bytes after a block of size 13 alloc'd
==27205==    at 0x4C2DB8F: malloc (vg_replace_malloc.c:299)
==27205==    by 0x545D489: strdup (strdup.c:42)
==27205==    by 0x40D4E0: split_on_colon(char const*, char**, unsigned long*) (main.cpp:118)
==27205==    by 0x4024A0: parse_argv (main.cpp:465)
==27205==    by 0x4024A0: main (main.cpp:790)
==27205== 
Open UART [4] /dev/ttyACM0 *
UART [4] speed = 115200
Open TCP [5] 0.0.0.0:5760 *
Logging target system_id=1 on 00003-2017-06-30_13-23-34.ulg
**27205** *** memcpy_chk: buffer overflow detected ***: program terminated
==27205==    at 0x4C3085C: VALGRIND_PRINTF_BACKTRACE (valgrind.h:6816)
==27205==    by 0x4C3544A: __memcpy_chk (vg_replace_strmem.c:1564)
==27205==    by 0x412B07: memcpy (string3.h:53)
==27205==    by 0x412B07: ULog::write_msg(buffer const*) (ulog.cpp:143)
==27205==    by 0x40E2D5: Mainloop::write_msg(Endpoint*, buffer const*) (mainloop.cpp:122)
==27205==    by 0x40E3B7: Mainloop::route_msg(buffer*, int, int, int, int) (mainloop.cpp:143)
==27205==    by 0x40A6BB: Endpoint::handle_read() (endpoint.cpp:77)
==27205==    by 0x40F61D: Mainloop::loop() (mainloop.cpp:282)
==27205==    by 0x40255E: main (main.cpp:806)
==27205== 
==27205== HEAP SUMMARY:
==27205==     in use at exit: 95,108 bytes in 18 blocks
==27205==   total heap usage: 38 allocs, 20 frees, 134,529 bytes allocated
==27205== 
==27205== LEAK SUMMARY:
==27205==    definitely lost: 0 bytes in 0 blocks
==27205==    indirectly lost: 0 bytes in 0 blocks
==27205==      possibly lost: 0 bytes in 0 blocks
==27205==    still reachable: 95,108 bytes in 18 blocks
==27205==         suppressed: 0 bytes in 0 blocks
==27205== Rerun with --leak-check=full to see details of leaked memory
==27205== 
==27205== For counts of detected and suppressed errors, rerun with: -v
==27205== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)

nkhoit, 2017-06-30:
Both of these are running PX4 on Stable 1.6.3 on a pixracer. Is there anything on the PX4 side that could be different?

zehortigoza, 2017-06-30:
@nkhoit This should fix the buffer overflow that you pointed out when running on your X86 machine. https://github.com/01org/mavlink-router/pull/102

zehortigoza, 2017-06-30:
Added another patch to fix the 'Invalid read of size 1'.

nkhoit, 2017-06-30:
I tried your branch. It didn't fix the buffer overflow in particular.
It did fix the problem with the whole colon baudrate thing. Here's an exact valgrind on my x86 machine connecting to a Pixracer.
valgrind ./mavlink-routerd -l ~ /dev/ttyACM0
==31670== Memcheck, a memory error detector
==31670== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==31670== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
==31670== Command: ./mavlink-routerd -l /home/nkhoit /dev/ttyACM0
==31670== 
Open UART [4] /dev/ttyACM0 *
UART [4] speed = 115200
Open TCP [5] 0.0.0.0:5760 *
Logging target system_id=1 on 00005-2017-06-30_16-08-56.ulg
**31670** *** memcpy_chk: buffer overflow detected ***: program terminated
==31670==    at 0x4C3085C: VALGRIND_PRINTF_BACKTRACE (valgrind.h:6816)
==31670==    by 0x4C3544A: __memcpy_chk (vg_replace_strmem.c:1564)
==31670==    by 0x412F4C: memcpy (string3.h:53)
==31670==    by 0x412F4C: ULog::write_msg(buffer const*) (ulog.cpp:152)
==31670==    by 0x40E2E5: Mainloop::write_msg(Endpoint*, buffer const*) (mainloop.cpp:122)
==31670==    by 0x40E3C7: Mainloop::route_msg(buffer*, int, int, int, int) (mainloop.cpp:143)
==31670==    by 0x40A75B: Endpoint::handle_read() (endpoint.cpp:77)
==31670==    by 0x40F62D: Mainloop::loop() (mainloop.cpp:282)
==31670==    by 0x40257B: main (main.cpp:807)
==31670== 
==31670== HEAP SUMMARY:
==31670==     in use at exit: 95,108 bytes in 18 blocks
==31670==   total heap usage: 38 allocs, 20 frees, 134,529 bytes allocated
==31670== 
==31670== LEAK SUMMARY:
==31670==    definitely lost: 0 bytes in 0 blocks
==31670==    indirectly lost: 0 bytes in 0 blocks
==31670==      possibly lost: 0 bytes in 0 blocks
==31670==    still reachable: 95,108 bytes in 18 blocks
==31670==         suppressed: 0 bytes in 0 blocks
==31670== Rerun with --leak-check=full to see details of leaked memory
==31670== 
==31670== For counts of detected and suppressed errors, rerun with: -v
==31670== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)

nkhoit, 2017-06-30:
Fixed in #102

closed 2017-06-30 by nkhoit