mavlink-router/mavlink-router #107: duplicated uorb topics

ChristophTobler, 2017-07-18:
When I run the optical flow app (https://github.com/zehortigoza/aero-optical-flow/tree/without-bmi) on the aero and publish the messages at 15Hz, I see them in PX4 (uorb top in mavlink_shell) at ~26Hz.
```
TOPIC NAME                    INST #SUB #MSG #LOST #QSIZE
optical_flow                     0    4   25    52 1
```
With `listener optical_flow 10`, I can see that some messages are duplicated (see timestamps in [output](http://pasted.co/325b34e6))

@zehortigoza FYI

lucasdemarchi, 2017-07-18:
Humn... Do you have any reason to believe it's a problem in mavlink-router?  I'd guess it's much more likely to be something in the aero-optical-flow repository.

ChristophTobler, 2017-07-18:
Could be, yes. I just verified that https://github.com/zehortigoza/aero-optical-flow/blob/without-bmi/src/main.cpp#L238 runs at ~15Hz. But @zehortigoza has better insight on that.

ChristophTobler, 2017-07-26:
@lucasdemarchi We're pretty certain that this is a mavlink-router issue. Two more indications:
* when running a mavlink shell and pressing enter it behaves like pressing enter twice
* run `mavlink status` in a mavlink shell and you will see that rx is identical to tx even though we are not sending anything. 

```
GCS heartbeat:	550067 us ago
	mavlink chan: #1
	type:		GENERIC LINK OR RADIO
	flow control:	ON
	rates:
	tx: 7.088 kB/s
	txerr: 0.000 kB/s
	rx: 7.088 kB/s
	rate mult: 1.000
	accepting commands: YES
	MAVLink version: 2
	transport protocol: serial (/dev/ttyS1 @460800)
```

Seems like mavlink-router is reflecting the messages.

lucasdemarchi, 2017-07-26:
I'm trying to reproduce this. I'm writing below how I'm testing, let me know if I am missing any step you are doing.

Connect to Aero over wifi. Make sure mavlink-router is running:

``` sh
ps -ef  | grep mavlink-router
```

csd is running as well by default but AFAIR it's not sending any mavlink packet on the released versions (and it didn't change anything here). Anyway, you can stop it:

``` sh
systemctl stop csd
```

On my host computer I start mavlink_shell:

``` sh
./Tools/mavlink_shell 0.0.0.0:14550
```

This is a replay of my shell:
https://asciinema.org/a/5hRVmlzNxWAfr5Cdg9og8Py81

It's reasonable to have a minimal rx since mavlink_shell itself is sending a heartbeat and I'm also typing on the shell.  The result here seems very different from what you are getting.  Did I miss any step you are doing?

ChristophTobler, 2017-07-27:
You are right! I think the issue was that I ran the mavlink-router with something like `mavlink-routerd -e 10.20.21.14:14550 -e 127.0.0.1:14550` (10.20.21.xx is the local network which my computer and the Aero uses) and I already had a main.conf that also streamed over usb (which was plugged in). My bad...

Is there maybe a way to protect against that?

NaterGator, 2017-08-14:
I can confirm / am absolutely seeing this. My fix was to add another parameter to route_msg that is just an Endpoint src pointer. Pass `this` in from the Endpoint when calling route_msg and in the body of route_msg skip endpoint matching the src pointer.

lucasdemarchi, 2017-08-14:
@NaterGator is this the same?   @ChristophTobler's issue was explained above as having 2 endpoints connected to the same place.

NaterGator, 2017-08-14:
The behavior I am seeing is even with a single UART and the default TCP endpoint mavlink-router reflects everything that comes in the UART back out it.

zehortigoza, 2017-08-14:
Looking at the code https://github.com/01org/mavlink-router/blob/master/src/mavlink-router/endpoint.cpp#L311 the messages are being reflected if a message without target is sent and the endpoint have more than one system id + component id.

The fix https://github.com/01org/mavlink-router/pull/113 would fix that but would also break any communication between systems in the same endpoint(not sure if this is real usage scenario) and maybe a better fix would be just remove '&& _sys_comp_ids.size() == 1' https://github.com/01org/mavlink-router/blob/master/src/mavlink-router/endpoint.cpp#L324

NaterGator, 2017-08-14:
I can't imagine any scenario where any of the current endpoints would break with #113. I'm not as strongly convicted about the fix in accept_msg, but if someone has a case where an offboard system is generating proxy messages for another component fixing there would break that functionality.

lucasdemarchi, 2017-08-14:
@NaterGator I think it's likely there's another bug triggering this one.  Do you really have more than one component in one endpoint ... this is something valid, but from what I've seen it's usually a bug in modules sending messages  (and AFAIR @zehortigoza fixed a few in PX4).

NaterGator, 2017-08-14:
@lucasdemarchi, no I do not... 

My setup is a px4 autopilot with an SI1000 radio for telemetry. The px4 autopilot is hosted on a TX1 and uses the companion USART for communication. On the TX1 I have a computer vision application running and accepting/generating mavlink messages (component ID 100). On the PX4 stack I have forwarding turned on in the mavlink instances because I want GCS commands coming from the ground to reach the application running on the Tegra and vice-versa. 

All of this works great, and I suspect what you are referring to the SI1000 telemetry updates propagating through the mavlink forwarding and appearing as multiple components on the same endpoint. Regardless, that should absolutely still be a supported use-case.

We have the following endpoint types, right?
- UDP
- TCP
- UART
- Log

Can you explain to me how it would ever make sense to perform loopback communication on these endpoints? I just can't envisage a use case where that behavior makes sense. In the UDP/TCP case it is an echo server, UART it is equivalent to wiring RX and TX together...

lucasdemarchi, 2017-08-14:
> @lucasdemarchi, no I do not...

From the rest of the answer... this should actually be "Yes, I do"?  In your case you are telling the flight stack to route messages from GCS (that has its own sysid + compid) through this endpoint. This endpoint is the same endpoint used to talk with fight stack and also GCS.

> Can you explain to me how it would ever make sense to perform loopback communication on these endpoints?

I don't think there's a case in which the loopback is valid or couldn't be replaced by something else.  We actually check this by comparing the src sysid/compid of that endpoint. What use cases do you see that fixing the check for src sysid/compid would not be sufficient?

lucasdemarchi, 2017-10-18:
Fixed

closed 2017-10-18 by lucasdemarchi