mavlink-router/mavlink-router #110: Mavlink router does not fall back to broadcast after the first connection times out

LorenzMeier, 2017-08-04:
I'll try to get better docs to http://mavlink.io, there is not enough or clear documentation on the discovery scheme.

If you connect with a GCS, disconnect, then reconnect with a different GCS mavlink router does not fall back to broadcast (at least according to our testing). It would be great to add this, as its consistent with our discovery scheme.

It should also stop sending heartbeats to broadcast once it is connected to a host to (I have not verified if that is the case right now or not, this is merely a general remark).

lucasdemarchi, 2017-08-04:
> If you connect with a GCS, disconnect, then reconnect with a different GCS mavlink router does not fall back to broadcast (at least according to our testing).

Yep. I'll add the label bug here. I always intended to implement something like that but lacked information about it.

What's the timeout? How to handle the case in which the same GCS should be used? I.e. the drone simply went too far for a long time and returned. The client side (gcs or others) may not be listening on broadcast anymore).

Or if one want to try a PR should be straightforward.

jbeyerstedt, 2021-12-09:
This issue is very old, so mavlink router might have changed in the meantime. Because I don't really understand what the issue is. (Especially, if "broadcast" means IP broadcast or mavlink message broadcast.)

An UDP client mode endpoint will just send messages to a port and does not know (and care), if anybody is listening. So this works perfectly fine sending data to the default UDP port of QGC. As soon as the GCS has sent a message back to mavlink-router on that socket (like parameter requests), it'll get these targeted answers, as well (and not just mavlink broadcast messages).

The same should be true for UDP connections from the GCS to mavlink-router (server mode UDP endpoint). Obviously mavlink-router is only able to send data through the socket as soon as a client has connected (sent data) once. So the GCS has to start with it's heartbeat.  
A TCP connection to the `TcpServerPort` works even better.

What am I missing?

lucasdemarchi, 2021-12-09:
@jbeyerstedt  the problem is that once we lock in to a specific ip:port (due to have received a packet from it), we never release it falling back to "sending packets to broadcast address, waiting for someone to connect". And by "connect" here it is basically "did we receive a message in endpoint in the last X seconds?".  This also assumes all endpoints send heartbeat.

lucasdemarchi, 2021-12-09:
Documentation added since this issue was created: http://mavlink.io/en/services/heartbeat.html

jbeyerstedt, 2021-12-10:
What endpoint type are you talking about? Because it can't be UDP Client mode (will always send to a fixed address), TCP client (will re-connect TCP, if not disabled) and TCP server (no client, no connection). So the only remaining type is UDP Server mode, which does not send any data until somebody connected. That's why I don't understand where it shall fall back to when it never has sent broadcast packets in the first place.

The heartbeat protocol specification also doesn't state anything about IP broadcast packets, just that the heartbeat message is a MAVLink broadcast (since there is no target sysID and compID). So mavlink-router will forward this message to all endpoint (unless it isn't filtered).

While writing this, I had an idea for one configuration, which won't work as expected: An UDP "normal mode" endpoint with a broadcast IPv4/ multicast IPv6 address. Because of [src/endpoint.cpp#L913](https://github.com/mavlink-router/mavlink-router/blob/1cc5529abffde7c41f87ff23e54ad2a09b4920b7/src/endpoint.cpp#L913), no message will be sent unless somebody has connected. But the GCS can't connect in the first place, because it doesn't know the port on mavlink-router's side, because mavlink-router doesn't send anything.

So in the end, I still don't understand in which configuration you would expect an IP broadcast. Sorry for asking in so much detail.

lucasdemarchi, 2021-12-10:
@jbeyerstedt it is udp client. Just pass a broadcast address.

```
int UdpEndpoint::open(const char *ip, unsigned long port, bool server)
{
...
        if (setsockopt(fd, SOL_SOCKET, SO_BROADCAST, &broadcast_val, sizeof(broadcast_val)) != 0) {
...
}
```

We use recvfrom() in _read_msg():

```
ssize_t UdpEndpoint::_read_msg(uint8_t *buf, size_t len)
{
...
    r = ::recvfrom(fd, buf, len, 0, sock, &addrlen);
...
}
```

So we will "lock" to the first client that sends something as reply to the messages we started broadcasting. Just tested this with ArduPilot + mavlink-router + QGC. It still works.

ArduPilot sends packets to port 10000
```
$ sudo ./build/sitl/bin/arducopter --model quad --uartA udpclient:127.0.0.1:10000
```

mavlink-router routes what is received in port 10000 to port 5000 with a broadcast address:
```
$ sudo ./build/src/mavlink-routerd  -e 127.0.0.255:5000 127.0.0.1:10000
```

QGC is configured to listen on port 5000.

This is the tcpdump for the moment QGC starts responding. Notice that mavlink-router then switches from 127.0.0.255 to 127.0.0.1 due to the lines I mentioned above.

```
11:36:20.721831 IP 127.0.0.1.51570 > 127.0.0.1.10000: UDP, length 17
11:36:20.721918 IP 127.0.0.1.53482 > 127.0.0.255.5000: UDP, length 17
11:36:21.720035 IP 127.0.0.1.51570 > 127.0.0.1.10000: UDP, length 17
11:36:21.720099 IP 127.0.0.1.53482 > 127.0.0.255.5000: UDP, length 17
11:36:22.720573 IP 127.0.0.1.51570 > 127.0.0.1.10000: UDP, length 17
11:36:22.720657 IP 127.0.0.1.53482 > 127.0.0.255.5000: UDP, length 17
11:36:23.721916 IP 127.0.0.1.51570 > 127.0.0.1.10000: UDP, length 17
11:36:23.721987 IP 127.0.0.1.53482 > 127.0.0.255.5000: UDP, length 17
11:36:23.724196 IP 127.0.0.1.5000 > 127.0.0.1.53482: UDP, length 41
11:36:23.724259 IP 127.0.0.1.10000 > 127.0.0.1.51570: UDP, length 41
11:36:23.727822 IP 127.0.0.1.51570 > 127.0.0.1.10000: UDP, length 11
11:36:23.727830 IP 127.0.0.1.51570 > 127.0.0.1.10000: UDP, length 68
11:36:23.727837 IP 127.0.0.1.53482 > 127.0.0.1.5000: UDP, length 11
11:36:23.727843 IP 127.0.0.1.53482 > 127.0.0.1.5000: UDP, length 68
```
Which means we will "lock" on that ip:port forever. I believe the issue @LorenzMeier  is mentioning is that we never switch back to the broadcast address if we stop receiving data from that endpoint.  Since it's udpclient, this is an issue if the 2 GCSs are actually in different machines, so the IP may be different.

jbeyerstedt, 2021-12-11:
Thanks, I'll add this to my PRs regarding the specification of the routing behavior (describing the current behavior in the first PR and discussing the intended behavior in an upcoming PR).

closed 2022-01-15 by lucasdemarchi

stevenjowens, 2024-02-15:
> What endpoint type are you talking about? Because it can't be UDP Client mode (will always send to a fixed address), TCP client (will re-connect TCP, if not disabled) and TCP server (no client, no [...]

This is one of the more useful comments I've found about the different modes.  Also the comment about heartbeats was useful.  Perhaps I'm just looking in the wrong places, but can you point me at something that goes into these in more detail?

jbeyerstedt, 2024-02-16:
The modes of the different endpoint types are described in the Readme: https://github.com/mavlink-router/mavlink-router/blob/master/README.md#endpoints. If something is missing there, please open a new issue or propose some changes through a pull request.

PR #325 (2022-01-15): UdpClient: switch back to broadcast on inactivity
More details on last commit message and on !10.

Fix: #110
commit a6d3a48d (2022-01-15): endpoint: udp: switch back to broadcast after a timeout

When we configure a UdpClient with a broadcast address, it has a special
behavior that it will send the packets as broadcast, but it will "lock
in" the first client that responds on our IP:ADDRESS.

However, after a client (e.g. a GCS) disconnects, it's useful to switch
back to broadcast because of the following scenarios:
1) User may initially connect with a computer and later decide to switch
   to a GCS in a tablet/phone
2) Client got disconnect from wifi and when reconnecting got another IP
3) Switch from one GCS to another.

So, if  [... cut]