mavlink-router/mavlink-router #112: Possible to access binlog in realtime?

fnoop, 2017-08-13:
As well as saving binlog to file, it would be great to access and process the data in realtime for analysis purposes.  Is this possible with mavlink-router, or it only can save to file?

lucasdemarchi, 2017-08-13:
Right now it only saves to a file. How do you plan to use it in realtime?

fnoop, 2017-08-13:
I'd like to try and extend pymavlink DFReader to use realtime logs, and pipe the data into InfluxDB in realtime to be used by eg. grafana front-end:
https://discuss.ardupilot.org/t/maverick-mavlink-dataflash-web-log-analysis-graphing/18767
https://github.com/fnoop/maverick/issues/500

At the moment, Maverick graphing uses inotify to watch for binlogs being written by mavlink-router to be closed, then when closed it processes them and imports the data into Influxdb.  I'd like this to happen in realtime.  I could write a simple python mechanism that requests mavlink_remote_log data like mavlink-router, but that would duplicate what mavlink-router is already doing and might conflict.

lucasdemarchi, 2017-08-14:
We open the file and only ever append to it. So in theory you should open the file in RO and just keep getting the data.  If that doesn't work we may add a pipe and send log there (in addition/replacement to saving it to a file).

lucasdemarchi, 2017-08-14:
In other words it may work right now as is, otherwise it would be a good feature to add.

fnoop, 2017-08-15:
The trouble with that approach (which is a good idea), is that DFReader opens and reads the entire file, then rewinds to the start.  The reason it does this is to scan for GPS time, which it uses to synchronise with the FC time-since-boot time.  The GPS time only appears after GPS lock is obtained some way through the log file.
I'll have another look at DFReader and see if it's possible to hold the file open somehow rather than read it all in at once.

lucasdemarchi, 2017-08-15:
If all it does is read and seek, you'd still be fine

fnoop, 2017-08-15:
Cool thanks, I'll close this until I have a go at it. If I have any problems from the mavlink-router end I'll reopen.

closed 2017-08-15 by fnoop