mavlink-router/mavlink-router #117: Virtual Port

terial, 2017-09-01:
This may be a dumb question and due to still learning how a non-wondows OS works but will mavlink-router support routing to a virtual port created by socat? I tried setting up the config file but did not get it to work.

Thanks

zehortigoza, 2017-09-06:
I never used a socat virtual serial port but if you can open screen/putty or any other tool that handles serial ports I guess mavlink-router will also be able to handle it.
I can't think in any reason why you would need a virtual serial port, what is your use case?

terial, 2017-09-07:
I am using wifi broadcast for video transmission. There are two instances of it running on the transmittter (RPI3) on the uav. One for the video transmission and another for the mavlink telemetry. I need a way to split the mavlink data on /dev/ttyAMA0 to the second instance. I am trying to get the mavlink data into a named pipe/fifo and then send that data to second tx  application. The command I think would then be something like this. Cat fifo1 | /opt/wifibroadcast/tx ..... this is a new learning experience for me so maybe I am not going about it the right way. Thanks for your time and help though

lucasdemarchi, 2017-09-11:
What's the interface you have with wifibroadcast? From your comment it seems you only have a pipe?  Currently we support UART, UDP and TCP. Adding named pipe could be something to think about.

Using socat should work if you are using either UDP-LISTEN and added the port on mavlink-router configuration or using TCP and made socat connect to port 5760  ( I haven't tested this though.... let me know if it works).

terial, 2017-09-28:
I am not sure what you mean by interface... to be honest getting all this running is a new challenging experience. The wifibroadcast OSD is looking for a serial port to send/receive mavlink data but currently the serial is being used by mavlink router so I am under the impression I need to split/route the stream.

terial, 2017-10-29:
Hey @lucasdemarchi I recently have been trying again to get this to work but seem to hit a block with everything I try. using socat to listen to TCP port 5760 using this command "socat TCP-LISTEN:5760 PTY,link=/dev/vs0,rawer,wait-slave" gives me this error "2017/10/29 06:14:46 socat[1489] E bind(5, {AF=2 0.0.0.0:5760}, 16): Address already in use" I also tried using UDP-LISTEN but got the same error. Any suggestions?

terial, 2017-10-29:
I got it figured out with socat. The command was 
"socat pty,link=$FC_TELEMETRY_SERIALPORT,raw,echo=0,waitslave tcp:127.0.0.1:5760"
and in the mavlink-router config i had the udp type to broadcast but on normal it seems to work. Now using cat i can see the mavlink data stream.

closed 2017-10-29 by terial