mavlink-router/mavlink-router #121: Download flightstack logs using mavlink-router

kevin-george, 2017-10-19:
We are running mavros 0.21 and wish to stream the flightstack logs using mavlink-router over udp. The commands we used were,

`mavlink-routerd -e 127.0.0.1:14550 -l /home/drone/logs /dev/pixhawk:921600`
/dev/pixhawk is the device we mapped using udev rules

`roslaunch mavros px4.launch fcu_url:=udp://@127.0.0.1:14550`

The output of **mavlink-routerd** command was
```
Could not open conf file '/etc/mavlink-router/main.conf' (No such file or directory)
Open UART [4] /dev/pixhawk *
UART [4] speed = 921600
Open UDP [5] 127.0.0.1:14550
Open TCP [6] 0.0.0.0:5760 *
```

The output of **examples/heartbeat-print.py** was
```
HEARTBEAT {type : 14, autopilot : 12, base_mode : 81, custom_mode : 458752, system_status : 3, mavlink_version : 3}
HEARTBEAT {type : 14, autopilot : 12, base_mode : 81, custom_mode : 458752, system_status : 3, mavlink_version : 3}
HEARTBEAT {type : 14, autopilot : 12, base_mode : 81, custom_mode : 458752, system_status : 3, mavlink_version : 3}
```

But when we launch mavros, it doesnt receive anything over the fcu_url.
The output of rostopic echo /mavros/state
```
header: 
  seq: 0
  stamp: 
    secs: 1508442258
    nsecs: 642219235
  frame_id: ''
connected: False
armed: False
guided: False
mode: ''
system_status: 0
---
```

Any help in setting this up would be appreciated! :)

lucasdemarchi, 2017-10-19:
Regarding the logging, have you checked https://dev.px4.io/en/log/logging.html ?

zehortigoza, 2017-10-19:
Try to run mavlink-routerd with "-r" so you can check if mavlink-router is sending and receiving packages in the UART and UDP ports but as it is working with the heartbeat-print.py I guess the problem lives in the mavros side.

kevin-george, 2017-10-19:
Yes, the problem we saw was that both mavros and the log streaming using pymavlink(using script mavlink_ulog_streaming.py) was not working as they both try to bind to the same serial port. 

A possible solution was to use mavlink-router as a proxy and have both mavros and mavlink_ulog_streaming.py talk to it instead. The latter(mavlink_ulog_streaming.py) works fine(generates logs) but the former(mavros) isn't able to communicate with mavlink-router using the commands seen above.

Am I missing something?

lucasdemarchi, 2017-10-19:
I successfully used mavros some time ago with mavlink-router. I don't remember the command exactly but I can check.

kevin-george, 2017-10-19:
@zehortigoza 
I did run it with -r. Here's part of the output
```
Endpoint AutoLog [-1] {
	Received messages {
		CRC error: 0 0% 0KBytes
		Sequence lost: 0 0%
		Handled: 0 0KBytes
		Total: 0
	}
	Transmitted messages {
		Total: 0 0KBytes
	}
}
Endpoint UART [4] {
	Received messages {
		CRC error: 1603 2% 245KBytes
		Sequence lost: 69749 93%
		Handled: 3179 331KBytes
		Total: 74531
	}
	Transmitted messages {
		Total: 263 43KBytes
	}
}
Endpoint UDP [5] {
	Received messages {
		CRC error: 0 0% 0KBytes
		Sequence lost: 0 0%
		Handled: 0 0KBytes
		Total: 0
	}
	Transmitted messages {
		Total: 2377 133KBytes
	}
}
```

This shows that it's reading data via the UART and transmitting over UDP but mavros isn't playing nice :(

zehortigoza, 2017-10-19:
`Sequence lost: 69749 93%`
Well there is something wrong in your setup, check the baudrate again of both ends again.

Looks mavros is connected and transmitting data to mavlink-router so the problem can be the above.

zehortigoza, 2017-10-19:
Another important thing is the mavlink-router the only user of the UART? if no that can explain this problem.

lucasdemarchi, 2017-10-19:
> roslaunch mavros px4.launch fcu_url:=udp://@127.0.0.1:14550

from [mavros' README](https://github.com/mavlink/mavros/blob/master/mavros/README.md#connection-url), this part seems wrong. It should be `udp://127.0.0.1:14550`

kevin-george, 2017-10-19:
@lucasdemarchi 
I followed this schema present in the [README](https://github.com/mavlink/mavros/blob/master/mavros/README.md#connection-url)
`UDP: udp://[bind_host][:port]@[remote_host[:port]][/?ids=sysid,compid]`  
If i use `udp://127.0.0.1:14550`, it complains saying 
```
[ INFO] [1508449754.932221703]: FCU URL: udp://127.0.0.1:14550
[ERROR] [1508449754.932335450]: UDP URL should contain @!
```

@zehortigoza 
I double-checked the baudrate. Is there anything else I can check?
Right now, mavlink-router is the only user of the serial port mapped to /dev/pixhawk.

lucasdemarchi, 2017-10-19:
@kevin-george than add a `@` to the end like `udp://127.0.0.1:14550@` since mavros should bind to the port.

kevin-george, 2017-10-31:
Sorry for the delay in getting back.

Thanks @lucasdemarchi That fixed the problem! The formatting of the UDP url was confusing :)
We were able to control multiple pixhawks via QGC and MAVROS.

closed 2017-10-31 by kevin-george