mavlink-router/mavlink-router #122: How to open a certain port on the Intel compute board running Yocto

RohanKurane, 2017-10-31:
Hi @zehortigoza @lucasdemarchi 
Looking at the examples here, looks like ports 5760 and 14450 are been used for MAVLINK messages. When I try to connect to a socket using tcp at port 5760 I get a ECONNREFUSED. How can I open a tcp connection from within my application that is running on the compute board to the flight controller i.e at the same ip address. I used 192.168.8.1 and 127.0.0.1 both with port 5760.

zehortigoza, 2017-10-31:
There is tested examples of how connect to TCP and UDP port in python and how to connect to TCP port using C/C++ here https://github.com/01org/mavlink-router/tree/master/examples

RohanKurane, 2017-10-31:
That is what I was referring to when I made my comment above. I was looking at px4-offboard-mode.cpp. How can I make sure port 5760 is open for listening ? I can try the UDP connection and see if that works.

zehortigoza, 2017-10-31:
You can add a printf() to handle_new_message() to check that it is receiving msgs.

RohanKurane, 2017-10-31:
Without setting up the connection how am I going to read any message ?  Dosent mavlink_parse_char() need to read a message from the socket connection ?

zehortigoza, 2017-11-01:
The example works, if it is refusing to connect to TCP 5760 port there is something wrong in your setup.
Check if mavlink-router is running, check the mavlink-router configuration file it should have the TCP port entry + UART entry.

RohanKurane, 2017-11-01:
I ran /etc/init.d/mavlink-routerd.sh start

This was by default in /etc/mavlink-router/main.conf.  Is this correct ?

[General]
TcpServerPort=5760
ReportStats=false
Log=/var/lib/mavlink-router/

[UartEndpoint uart]
Device = /dev/ttyS1
Baud = 460800

[UdpEndpoint wifi]
Mode = Normal
Address = 192.168.8.255

[UdpEndpoint ethernet-over-usb]
Mode = Normal
Address = 192.168.7.255
~

FYI..
The mavlink-routerd.sh script has 57600 for baud rate

# Functions to do individual actions
start(){
         nohup /usr/bin/mavlink-routerd -b 57600 \
                2>/var/log/mavlink-routerd.log \
                >/dev/null &

zehortigoza, 2017-11-01:
You should definitely update the Yocto image in your board, not 100% but I think you are running 1.3.1V and if I remember correctly in this image the Wifi and usb-over-ethernet IPs are different, please confirm that to me. Also run mavlink-router with "-r" it should print the statistics of each endpoint.
QGround Control can connect to the drone? it works?

RohanKurane, 2017-11-01:
@zehortigoza 

root@intel-aero:~# aero-get-version.py

BIOS_VERSION = Aero-01.00.12_Prod
OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.3.1 (krogoth)"
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xa1

QGC can connect to the drone. I can stream live video taken from the realsense camera and send it over using GStreamer to my laptop and view it using QGC. 

You mean run the mavlink-routerd.sh script with the -r option ?  It gives me this -
root@intel-aero:~# /etc/init.d/mavlink-routerd.sh -r
Usage: mavlink-routerd { start | stop | status | restart }

I am reading the README in
https://github.com/01org/mavlink-router/blob/master/README.md

root@intel-aero:~# git submodule update --init --recursive
-sh: git: command not found

Isnt all this built already on the compute board ? What do I have to do ? Yocto does not have git installed. Not sure how to install it.

RohanKurane, 2017-11-01:
Im also a bit confused..
The README is suppose to be done on my laptop I assume ?  I am running my app on the compute board at the moment and testing MAVLINK. I should be able to receive MAVLINK messages within my application thats running on the board. The TCP connection is failing to succeed.

zehortigoza, 2017-11-01:
> You mean run the mavlink-routerd.sh script with the -r option ? It gives me this -
> root@intel-aero:~# /etc/init.d/mavlink-routerd.sh -r
> Usage: mavlink-routerd { start | stop | status | restart }

Just run "mavlink-router" with the parameters needed you can look at the script( /etc/init.d/mavlink-routerd.sh) to check what is the other parameters.

> I am reading the README in
> https://github.com/01org/mavlink-router/blob/master/README.md
>
> root@intel-aero:~# git submodule update --init --recursive
> -sh: git: command not found

I mean update the Linux image, we are now in the version 1.6v: https://github.com/intel-aero/meta-intel-aero/wiki/02-Initial-setup#flashing

> Im also a bit confused..
> The README is suppose to be done on my laptop I assume ? I am running my app on the compute > board at the moment and testing MAVLINK. I should be able to receive MAVLINK [...]

With 127.0.0.1 it will only work when running inside of the board but using the regular IP it should work running outside of the board.

RohanKurane, 2017-11-01:
Has mavlink-routerd.sh been updated since 1.3 ? 
It dosent take anything except start/stop/restart/status on 1.3

root@intel-aero:~# vi /etc/init.d/mavlink-routerd.sh 
Usage: mavlink-routerd { start | stop | status | restart }

I will update to 1.6 today. 

My application is running inside the board. The PX4 is connected to the board. My application tries to set up the TCP connection using 127.0.0.1 port 5760. That fails with ECONNREFUSED. Once the connection is setup, I will be reading from the socket for any MAVLINK messages and process them within my application. 

So the only thing I am missing now is to upgrade the board to 1.6 ?

zehortigoza, 2017-11-01:
> Has mavlink-routerd.sh been updated since 1.3 ?
> It dosent take anything except start/stop/restart/status on 1.3
> 
> root@intel-aero:~# /etc/init.d/mavlink-routerd.sh
> Usage: mavlink-routerd { start | stop | status | restart }

There is no  /etc/init.d/mavlink-routerd.sh in the newer releases, now it uses systemd.

> So the only thing I am missing now is to upgrade the board to 1.6 ?

You should also update PX4 firmware, BIOS and FPGA firmware all that is covered by the wiki https://github.com/intel-aero/meta-intel-aero/wiki/02-Initial-setup

RohanKurane, 2017-11-01:
Okay. I will update all the software. 

One more question - After installing the latest software, and connecting the flight controller to the compute board, then setting up the connection with itself at 127.0.0.1 - I should just be able to read the file descriptor and see the MAVLINK messages come in (I will add printfs as well to see what they are). Is that correct ? Or do I need to run anything else to generate the MAVLINK messages that I read from the socket ? 
/etc/init.d/mavlink-router start on 1.3 release should be enough ? Or anything else ?

zehortigoza, 2017-11-01:
After you update everything you just need to copy one of the examples to aero and run it.
Output of px4-offboard-mode correctly running indoors:
```
root@intel-aero:~# ./px4-offboard-mode
Connecting to TCP: 127.0.0.1:5760
Connected
Waiting for:
        - heartbeat
        - basic extended info
        - home position(GPS fix)
Waiting for:
        - home position(GPS fix)
Waiting for:
        - home position(GPS fix)
Waiting for:
        - home position(GPS fix)
Waiting for:
        - home position(GPS fix)
```

Run: aero-get-version.py
Also will validate that TCP port is working because it get the PX4 version using TCP.
```
root@intel-aero:~# aero-get-version.py

BIOS_VERSION = Aero-01.00.13
OS_VERSION = Poky Aero (Intel Aero Linux Distro) 1.6.0 (pyro)"
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xc2
AeroFC firmware version = 1.6.5
```
This is the output that you should have after update everything.

RohanKurane, 2017-11-03:
Hi @zehortigoza 

For some reason setting up a TCP socket fails. I can create a UDP socket as is done in heartbeat-print.cpp. 
Once I connect to the socket and then try to read from it, I get a -1 (errno = 11 EAGAIN). I guess that means there is nothing to read from the socket. 

I believe the TELEM2 is now correctly connected to the board. Why am I not receiving any MAVLINK messages after connecting to the socket (I started /etc/init.d/mavlink-router start) ? I am still on 1.3 for some other reasons but I would think this should work there as well. 
 
I do the same thing that is in heartbeat-print.cpp
       char *ip = (char *)"127.0.0.1";
       int port = 14550;

       tcp_fd = socket(AF_INET, SOCK_DGRAM, 0);

       if (tcp_fd == -1) {
          printf("Could not create socket\n");
          return -1;
       }

       bzero(&sockaddr, sizeof(sockaddr));
       sockaddr.sin_family = AF_INET;
       sockaddr.sin_addr.s_addr = inet_addr(ip);
       sockaddr.sin_port = htons(port);

       printf("Connecting to TCP: %s:%i\n", ip, port);

       if (bind(tcp_fd, (struct sockaddr *)&sockaddr, sizeof(sockaddr)) == -1) {
           printf("Could not bind to %s:%d:%d", ip, port, errno);
           return -1;
       }

ssize_t n = read(tcp_fd, buf, sizeof(buf));
 
if (n == -1)
    {
         printf("Error reading socket: %d\n", errno);
         goto finish;
}

avinash-palleti, 2017-11-03:
> int port = 14550;

@RohanKurane By default mavlink-router is configured to TCP port 5760 unless you modified in main.conf file, Can you try changing port to 5760.

RohanKurane, 2017-11-03:
@avinash-palleti 
Looks like the -b argument in mavlink-routerd is not valid anymore. I took that out and it seemed to work. I can now setup a tcp connection to 127.0.0.1 at port 5760. 
Then when I read the socket I get some 280 bytes back. 
But mavlink_parse_char(MAVLINK_COMM_0, buf[i], msg, &status) returns 0. 

Why do you think that is ?

zehortigoza, 2017-11-03:
460800 is the baudrate that pixhawk telemetry port is running?

We have this guide https://github.com/intel-aero/meta-intel-aero/wiki/93-(References)-Pixhawk-FC that was already validated by other users.

RohanKurane, 2017-11-08:
@zehortigoza 
I changed the main.conf file to reflect that 57600 baud rate.
I keep getting either an error (EAGAIN = 11) when I read the tcp socket connection for MAVLINK messages. mavlink_router is started and the connection is successful (127.0.0.1:5760).

What am I missing here to see MAVLINK messages in the socket ?

zehortigoza, 2017-11-08:
If you can connect to the TCP port and when you try to read it you get EAGAIN it mean that mavlink-router is reading nothing from UART, in this same setup QGC is working? You can set ReportStats=true and check the statics about how many messages are read and write in each endpoint.

Saying AGAIN update your board and follow this: https://github.com/intel-aero/meta-intel-aero/wiki/93-(References)-Pixhawk-FC

RohanKurane, 2017-11-08:
My log file is full of these lines
(../git/ulog.cpp:157) MAV_CMD_LOGGING_START result(1) is different than accepted    
(../git/ulog.cpp:157) MAV_CMD_LOGGING_START result(1) is different than accepted        
....

zehortigoza, 2017-11-08:
Upload it to https://logs.px4.io/

RohanKurane, 2017-11-08:
I did. I have no idea what I am looking at :)

zehortigoza, 2017-11-09:
> I did. I have no idea what I am looking at :)
>
> Can you view it as well ?

If you share the link I can.

RohanKurane, 2017-11-09:
Here you go
https://logs.px4.io/plot_app?log=2cb35350-b96e-4df2-91a4-ba78e39a531d

zehortigoza, 2017-11-09:
The UART<->Mavlink Router communication is working.

Maybe this warnings are from your application send mavlink packages with the same sys/comp id as PX4?
ignoring CMD with same SYS/COMP (1/1) ID

RohanKurane, 2017-11-09:
I am really not sure what you mean. 
Looks like now I am getting some bytes back (20 bytes or so from the connection). Then when I call mavlink_parse_char for each byte in the 20 byte buffer, I get a segmentation fault at

x00007fc3a9f38bfe in mavlink_frame_char_buffer (rxmsg=0x7fc3aa747fc0 <mavlink_get_channel_buffer::m_mavlink_buffer>, 
    status=0x7fc3aa749200 <mavlink_get_channel_status::m_mavlink_status>, c=253 '\375', r_message=0x0, 
    r_mavlink_status=0x7fc3ae33b870) at /usr/include/mavlink_helpers.h:814
814		r_message->len = rxmsg->len; // Provide visibility on how far we are into current msg

Have you seen this before ?

zehortigoza, 2017-11-09:
Run one of the examples that runs over TCP, check their code and compare with yours.
I can not help you fix your code.

RohanKurane, 2017-11-09:
I understand. This is something in mavlink helper functions. Thats why I asked. But I will look.

RohanKurane, 2017-11-09:
@zehortigoza 
It looks better now. 
Question - when I parse the buffer, sometimes mavlink_parse_char fails to parse the byte correctly. Sometimes it parses it correctly and a message id is returned. 

Is this normal ?

RohanKurane, 2017-11-10:
@zehortigoza  
Can you or anyone else on the team comment on my last update above ? 
Sometimes mavlink_parse_char fails to parse the character in the buffer returned from the socket. Sometimes it passes and a message can be decoded. 
Is this normal ? 
Am I doing something wrong ?

zehortigoza, 2017-11-11:
This is beyond of what we give support here but look at the result parameter the field parse_state can give you the reason of why mavlink_parse_char() returned 0.

closed 2017-11-11 by zehortigoza