mavlink-router/mavlink-router #125: Mavlink router causes slow network when multiple WiFi clients are connected

compeoree, 2018-02-05:
**Environment**
- Aero RTF (Image version 1.6.1)
- Ubuntu Server 16.04 client, ssh (PC1)
- Android client, no terminal, only connect to Wifi (Mobile)
- Windows 10 client, Putty (PC2)

No video feeds or MAV endpoints (HTTP, VLC, QGroundControl, etc) were active. Only SSH sessions were used

**Steps to Reproduce**
1. Connect to Aero Wifi with PC1
2. Open SSH session to Aero
3. Run 'top' with update time of 0.1 seconds
4. Observe proper behavior of 'top' (screen refreshes are timely)
5. Connect second and/or third client to Aero Wifi (PC2 and/or Mobile)
6. Observe erratic timing of 'top' screen updates
7. Observe slow response of SSH shell in general

**Resolution 1**
Stop mavlink-router service ('systemctl stop mavlink-router')
Observe proper behavior of SSH when multiple clients are connected

**Resolution 2**
Disconnect second/third clients
Observe proper behavior of SSH 

System Log
[mavlink-router.log](https://github.com/intel/mavlink-router/files/1695935/mavlink-router.log)

jbeyerstedt, 2021-09-09:
Are all of your clients connected to each other via WiFi?

I have two ideas, what the issue could be:
1) Your WiFi is flooded with the UDP (or TCP) packets of the MAVLink connections. Or the WiFi router is overwhelmed.
2) The client running mavlink-router is too slow/ waiting for IO.
  - If the log files are written to an SD card, it might as well be a slow SD card. Because your log shows many newly created log files.
  - If that system is also the WiFi router, a performance issue with mavlink-router _might_ also affect the WiFi.

Currently I'm just guessing. But maybe this issue has fixed itself in the years, which have passed.

jbeyerstedt, 2022-02-27:
Closing this issue because it's quite old and has no activity. Please comment, if this still is an issue with the current master branch state and I'll re-open this issue.

closed 2022-02-27 by jbeyerstedt