mavlink-router/mavlink-router #126: Routing mavlink command traffic coming from Ground control to companion computer

myeabhishek, 2018-02-10:
Hi Team,

I am using pixhawk 2 with px4 connected to rpi with ROS. I am using QGC as my ground control software. I want to receive mavlink commands sent from qgc onto my companion computer. 

Can setting up mavlink_router help me with this? Can you please guide me on how to use this?

zehortigoza, 2018-04-13:
without any changes you will receive all the mavlink message without system id and component id target, to get the message targeting QGC you will need to hack into mavlink-router, most of your changes would be in: https://github.com/intel/mavlink-router/blob/master/src/mavlink-router/mainloop.cpp#L134

jbeyerstedt, 2021-01-27:
I'll close this because of inactivity for now. Just comment, if you have any updates.

closed 2021-01-27 by jbeyerstedt