mavlink-router/mavlink-router #127: Mavlink-router port 14560 to use HITL mode of PX4 is not working

trgiman, 2018-03-18:
Hi @lucasdemarchi I moved the issue here. I am trying to connect to port 14560 to /dev/ttyS1 of Yocto from my development workstation(Ubuntu) in order to be able to use HITL simulation (SYS_HITL in PX4 enabled). But am not getting any MavLink responses.

Any suggestions how to change /etc/mavlink-router/main.conf? I tried to add PORT = 14560,where appropriate according manual. But it seems, there is no change.

I have no issues using QGroundControl or ./mavlink_shell 0.0.0.0:14550 command when AeroRTF's WIfi or USB port is connected?

**main.conf file understanding:**

1. Yocto USB UDP connection assigned IP address to my Ubuntu workstation device (192.168.7.24)
2. Now I am trying to connect for instance with jMavSim to 192.168.7.2:14560 (Aero Board Yocto USB) 
3. In main.conf I created UDP connection with address= 192.168.7.2, mode=normal, port=14560
4. This should forward Mavlink Messages to AeroFC and AeroFC should respond to jMavSim. jMavSim will produce output:

Init MAVLink
MSG: Switched to ON hil state
MSG: [cmd] DISARMED by set mode command

lucasdemarchi, 2018-03-18:
If I understand correctly you are running jmavsim in your pc. Address in mavlink-router should contain that  address, not aero's.

It's easier if you copy and paste your config.

trgiman, 2018-03-18:
@lucasdemarchi . Yes I run jmavsim on ubuntu laptop. My current mavlink-router config on Yocto (Aero RTF) is as following

[General]
TcpServerPort=5760
ReportStats=false
Log=/var/lib/mavlink-router/

[UartEndpoint uart]
Device = /dev/ttyS1
Baud = 921600,460800

[UdpEndpoint wifi]
Mode = Normal
Address = 192.168.8.255

[UdpEndpoint ethernet-over-usb]
Mode = Normal
Address = 192.168.7.255
Port = 14560 (when I changed this param I can connect to NSH console with mavlink_shell 0.0.0.0:14560 using USB connection)

[UdpEndpoint csd]
Mode = Normal
Address = 0.0.0.0
Port = 80550
Mode = eavesdropping

When I run command
./jmavsim_run.sh 192.168.7.13:14560 (192.168.7.17 addressed assigned to USB connection of Ubuntu laptop by Yocto). I have got no connection to Aero FC....same results for 0.0.0.0 or 127.0.0.1

** My config of jmavsim_run.sh is:** 

#! /bin/bash

SCRIPT_DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
cd "$SCRIPT_DIR/jMAVSim"

udp_port=14560
extra_args=
baudrate=921600
device=""
ip="192.168.7.2"

ss -au command shows:

State      Recv-Q Send-Q            Local Address:Port                             Peer Address:Port                
UNCONN     0      0                             *:ipp                                         *:*                    
UNCONN     0      0                             *:mdns                                        *:*                    
UNCONN     0      0                             *:47184                                       *:*                    
UNCONN     0      0                    *%enp1s0f0:47736                                       *:*                    
UNCONN     0      0                     127.0.1.1:domain                                      *:*                    
UNCONN     0      0                             *:bootpc                                      *:*                    
UNCONN     0      0                             *:bootpc                                      *:*                    
UNCONN     0      0                            :::45589                                      :::*                    
UNCONN     0      0                            :::mdns                                       :::*                    
ESTAB      0      0           ::ffff:192.168.7.17:54564                      ::ffff:192.168.7.2:14560                
ESTAB      0      0              ::ffff:127.0.0.1:55161                        ::ffff:127.0.0.1:14550         

ESTAB is jMavsim connection. Should I add 192.168.7.17 into main.conf? It is dynamically assigned by Yocto to my Ubuntu USB interface.

avinash-palleti, 2018-03-19:
@trgiman Modifications to main.conf looks fine.
While you are running jmavsim, you need to use Aero's ip addresss.
Can you please try changing below command 
`./jmavsim_run.sh 192.168.7.13:14560 ` to `./jmavsim_run.sh 192.168.7.2:14560 ` (Usually ip address of Aero board on USB network is 192.168.7.2)

trgiman, 2018-03-19:
@avinash-palleti I changed ./jmavsim_run.sh 192.168.7.2:14560 . But there is no difference. My next steps are below. I want to eliminate if there is an problem with mavlink-router to Aero FC (PX4).

**Config of jMavsim is:**

* My config of jmavsim_run.sh is:**

#! /bin/bash

SCRIPT_DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
cd "$SCRIPT_DIR/jMAVSim"

udp_port=14560
extra_args=
baudrate=921600
device=""
ip="192.168.7.2" --> must be here otherwise Mavsime connects from localhost to localhost

**As a next step to resolve the issues i did following:**

1. ssh root@192.168.7.2 from my Ubuntu workstation to Aero RTF (Aeroboard Yocto)

2. netstat -aulp | grep "mavlink-routerd" - list all udp connection of mavlink-routerd

root@intel-aero:/etc/mavlink-router# netstat -aulp | grep "mavlink-routerd"
udp        0      0 *:39489                 *:*                                 356/mavlink-routerd
udp        0      0 *:15014                 *:*                                 356/mavlink-routerd
udp        0      0 *:54393                 *:*                                 356/mavlink-routerd

Also I noticed that last section of mavlink-routerd main.conf includes both modes normal and eavesdropping is this correct?

[UdpEndpoint csd]
Mode = Normal
Address = 0.0.0.0
Port = 80550
Mode = eavesdropping

trgiman, 2018-03-19:
@avinash-palleti I did further investigation and it seems, some MAVLINK data arrived to Aero FC. But than communication crashed. Please see below.Can you please suggest if this is wrong configuration of mavlink-router or is this related to jMavSim?

 1.ssh root@192.168.7.2 ---> AeroBoard Yocto Linux

 2. netstat -aulp | grep "mavlink-routerd"

udp 0 0 *:50506 : 341/mavlink-routerd
udp 0 0 *:41355 : 341/mavlink-routerd
udp 0 0 *:15014 : 341/mavlink-routerd

4. When I try run ./jmavsim_run.sh -q -i 192.168.7.2 -p 41355 -r 250 (on 41355 port or other 2 ports)

a)jMavsim writes **Init MAVLink.** Than it throws exception:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

Init MAVLink
Exception in Simulator.world.update() :
java.lang.IllegalArgumentException
at java.nio.Buffer.position(Buffer.java:244)
at me.drton.jmavlib.mavlink.MAVLinkMessage.(MAVLinkMessage.java:121)
at me.drton.jmavlib.mavlink.MAVLinkStream.read(MAVLinkStream.java:48)
at me.drton.jmavsim.UDPMavLinkPort.update(UDPMavLinkPort.java:139)
at me.drton.jmavsim.MAVLinkConnection.update(MAVLinkConnection.java:44)
at me.drton.jmavsim.World.update(World.java:33)
at me.drton.jmavsim.Simulator.run(Simulator.java:404)
at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
at java.lang.Thread.run(Thread.java:748)

a) NuttX PX4 nsh shell writes:

ERROR [sensors] Mag #0 fail: TIMEOUT!
ERROR [sensors] Accel #0 fail: TIMEOUT!
ERROR [sensors] Gyro #0 fail: TIMEOUT!

b) In nsh shell I run listener sensor_combined.....I can see that some data has been sent before Mavsim crashed...obviously uorb topics for sensors are silent after the communication is ceased

nsh> listener sensor_combined

TOPIC: sensor_combined instance 0 #1
timestamp: 1875614246
gyro_rad: 0.0175 0.0021 0.0208
gyro_integral_dt: 3981
accelerometer_m_s2: -0.0629 -0.0191 -9.8326
accelerometer_integral_dt: 3981
magnetometer_ga: 0.9034 0.0367 1.7795
baro_alt_meter: 487.9796
baro_temp_celcius: 0.0000

avinash-palleti, 2018-03-21:
As per your order of steps I understand that you are running below command on Aero board,
```4.When I try run ./jmavsim_run.sh -q -i 192.168.7.2 -p 41355 -r 250 ```
In that case ipaddress should be 0.0.0.0 and port number also should be 14550.

lucasdemarchi, 2018-09-11:
This seems something to report on either intel-aero's or px4's github.  If you think this is not the case, please let me know and I will double check what I read above.

closed 2018-09-11 by lucasdemarchi

trgiman, 2018-09-12:
Hi @lucasdemarchi. As you did mentioned. This issue has been solved within jMavsim project in March.Thanks.