mavlink-router/mavlink-router #128: Mavlink router tries to configure DTR/RTS and fails

estechnical, 2018-03-24:
I am trying to get an RK3328 based board to run mavlink router and using the onboard UART.

I have been able to test the UART with `cat` and `echo` to confirm data is send and received ok. This much is working. When I start mavlink router I get an error 

  Could not set DTR/RTS (Invalid argument)

and I cannot proceed.

Is there a way of starting this or does it require code changes to stop this from failing for my use case?

Many thanks,
Ed

estechnical, 2018-03-24:
I found the source of the problem, but I'm unsure what to do for my case.

https://github.com/intel/mavlink-router/blob/master/src/mavlink-router/endpoint.cpp#L540-L545

I suspect the port is not capable of DTR/RTS and this is why it's failing.

estechnical, 2018-03-24:
I have managed to make this work on the chosen board but expect my changes are not suitable for every use case.

https://github.com/estechnical/mavlink-router/tree/rock64

Is there a clean way it could be made to work? Eg a warning rather than a failure?

I'm happy to submit a PR with whatever changes you suggest - but I'm not sure why I needed to do this :)

Thanks,
Ed

zehortigoza, 2018-04-13:
maybe add a parameter to avoid configure DTR/RTS in the UART endpoints, leave the default option as configure it.

lucasdemarchi, 2018-09-20:
@estechnical see #141 - can you check if that fixes it for you?

PR #141 (2019-11-01): Revert "endpoint: set RTS/DTR state"
This reverts commit 940f0b8caf3cf61879c3cdfebd66a78eb127bd70.

I think this was probably one artficact of putting the uart in raw mode
rather than sane mode. We don't care about RTS/DTR and none of the
flight stacks we support are setting those.

Fix #128, but needs testing
commit 02cadb30 (2019-11-01): Revert "endpoint: set RTS/DTR state"

This reverts commit 940f0b8.

I think this was probably one artficact of putting the uart in raw mode
rather than sane mode. We don't care about RTS/DTR and none of the
flight stacks we support are setting those.

Fix #128, and also when we simulate a UART port via socat.
closed 2019-11-01 by lucasdemarchi