mavlink-router/mavlink-router #129: AeroRTF mavlink-router is loosing packet HITL simulation fails

trgiman, 2018-03-27:
Hi all. I connected jMavsim and as well sitl-gazebo in HITL mode to AeroFC. Unfortunately mavlink-router is loosing huge amounts of packets. So simulation especially with sitl-gazebo (very shaky, unusable due to dropped messages for actuator_output). jMavsim can be used but still similar problem. Could you please suggest the solution. It is Aero RTF and I need to run HITL simulation. 

**Mavlink-router config on yocto:**

[General]
TcpServerPort=0
ReportStats=false
Log=/var/lib/mavlink-router/

[UartEndpoint uart]
Device = /dev/ttyS1
Baud = 921600,460800

[UdpEndpoint ethernet-over-usb]
Mode = Normal
Address = 192.168.7.255
Port = 14560

**mavlink-routerd -r output (in this case jMavsim bu we need ti use Gazebo actually):**

Endpoint UDP [4] {
	Received messages {
		CRC error: 0 0% 0KBytes
		Sequence lost: 0 0%
		Handled: 60059 4432KBytes
		Total: 60059
	}
	Transmitted messages {
		Total: 117217 6847KBytes
	}
}
Endpoint UART [5] {
	Received messages {
		CRC error: 7550 1% 1308KBytes
		Sequence lost: 238141 61%
		Handled: 141087 12847KBytes
		Total: 386778
	}
	Transmitted messages {
		Total: 61079 4555KBytes
	}
}
Endpoint ULog [-1] {
	Received messages {
		CRC error: 0 0% 0KBytes
		Sequence lost: 0 0%
		Handled: 193 2KBytes
		Total: 193
	}
	Transmitted messages {
		Total: 22778 5968KBytes
	}

lucasdemarchi, 2018-03-27:
> Baud = 921600,460800

Baud is maximum 460800. Even then a 5~10% sequence lost may happen and need to be handled by the upper layer.  Disabling the log allows to have more bandwidth in the link, too.

zehortigoza, 2018-03-27:
Looks like there is a mismatch of baudrate, check what baudrate mavlink-router is detecting(921600 or 460800) and what is the baudrate of what you have in /dev/ttyS1.

trgiman, 2018-03-28:
Hi @lucasdemarchi @zehortigoza following are the results. Results are much better for jMavsim. But sitl_gazebo plugin is still big issue.

1. I disabled Log in main.config for mavlink-router
2. Running jMavsim and getting much better results compared to log above

Endpoint UDP [4] {
	Received messages {
		CRC error: 0 0% 0KBytes
		Sequence lost: 0 0%
		Handled: 41528 3064KBytes
		Total: 41528
	}
	Transmitted messages {
		Total: 96291 4544KBytes
	}
}
Endpoint UART [5] {
	Received messages {
		CRC error: 34 0% 1KBytes
		Sequence lost: 16051 14%
		Handled: 96291 4544KBytes
		Total: 112376
	}

3. But when I run sitl_gazebo plugin I have similar results for  Endpoint UART5 but for Endpoint UDP4 I have Sequence Lost 66%...

4. I checked and mavlink-routerd command confirms speed 921600 for /dev/tty1 and when I am in nuttx shell running mavlink status for /dev/tty1 it also confirms speed 921600....param show SYS_COMPANION command also shows 921600 which needs to be set for Gazebo.

Any Ideas for UDP4 endpoint please?

lucasdemarchi, 2018-03-28:
> But when I run sitl_gazebo plugin I have similar results for Endpoint UART5 but for Endpoint UDP4 I have Sequence Lost 66%...

UDP should not be losing any package.... Maybe you have conflicting sysids?

> I checked and mavlink-routerd command confirms speed 921600 for /dev/tty1 and when I am in nuttx shell running mavlink status for /dev/tty1 it also confirms speed 921600....param show SYS_COMPANION command also shows 921600 which needs to be set for Gazebo.

921600 is not sustainable for low error count, sorry. Try 460800... I never used the gazebo plugin, but I don't think there's a real reason not to work correctly with half the speed (I'm thinking theoretically on the data that should be exchanged). I'm still at lost why you are using HITL to start with. You are only introducing latency in the communication to the flight stack. Using sitl makes much more sense since you are not really using any sensor from the FC.

If you want to continue doing this and the gazebo plugin can't be fixed, then I suggest using another uart directly to the FC. You can connect a usb-serial to the telem port and increase its speed.

trgiman, 2018-03-28:
@lucasdemarchi Thanks for recommendations. 

1. I tried to decrease Gazebo world updates to 250HZ. Error rate felt down rapidly on udp4 to 24% on UART to 5%. Anyway gazebo_sitl still unusable

> UDP should not be losing any package.... Maybe you have conflicting sysids?
Do you mean sysids or component ids? jMavsim works ok with 0% errors 0% sequence lost in HITL on UDP4

> 921600 is not sustainable for low error count, sorry. Try 460800... I never used the gazebo plugin, but I don't think there's a real reason not to work correctly with half the speed (I'm thinking [...]

a)That is recommended setup for Gazebo at PX4 web page.Also default setup for SYS_COMPANION for Intel Aero RTF. I am a bit confused here. Should I change it in rcS and 4070_aerofc files?

b) target is to use "indoor gps" sensor connected to Aero FC and also to have 1:1 environment as much as possible? PX4 sitl is running POSIX versus AeroFC is running Nuttx. But will have to think about it again.

> If you want to continue doing this and the gazebo plugin can't be fixed, then I suggest using another uart directly to the FC. You can connect a usb-serial to the telem port and increase its speed.

I would like to do it. But Aero FC TELEM port has RXTX pins and according Intel Documentation it is named UART 5 and it seems the max speed is only 115200 bauds. I can use it successfully for nuttx boot console at speed 57600 bauds or direct there mavlink messages at 115200 bauds correct?I can not set it up for higher speeds.Right?If yes should I change the baud rates only in aerofc-v1/defconfig file?

lucasdemarchi, 2018-09-11:
There's nothing much mavlink-router can do here... that baudrate is not sustainable for that hardware.

closed 2018-09-11 by lucasdemarchi