mavlink-router/mavlink-router #134: [question] "unknown endpoints"

lamping7, 2018-07-25:
```
Open UDP [4] 127.0.0.1:14540  
Open UDP [5] 127.0.0.1:14550  
Open UART [6] /dev/ttySAC0 *
UART [6] speed = 921600
Open TCP [7] 0.0.0.0:5790 *
100 messages to unknown endpoints in the last 5 seconds
100 messages to unknown endpoints in the last 5 seconds
110 messages to unknown endpoints in the last 5 seconds
110 messages to unknown endpoints in the last 5 seconds
110 messages to unknown endpoints in the last 5 seconds
110 messages to unknown endpoints in the last 5 seconds
110 messages to unknown endpoints in the last 5 seconds
110 messages to unknown endpoints in the last 5 seconds
```
What does this message actually mean? I can receive some messages on the UDP endpoint, but when trying transfer a mission with MISSION_COUNT, I'm not receiving the MISSION_REQUEST and the above warning message appears.

zehortigoza, 2018-07-26:
It means that X messages coming from any endpoint had a target system id and component id that do not match with any endpoint connected to mavlink-router.
You are probably not setting target_system and target_component in MISSION_COUNT message, try to set to 1 and MAV_COMP_ID_ALL respectively also take a look at this example that handles mission messages: https://github.com/intel/mavlink-router/blob/master/examples/arm-authorizer.cpp

lamping7, 2018-07-26:
I am setting 1, 1. I can connect pymavlink directly to the flight controller serial port and everything works fine.

lamping7, 2018-07-26:
I'm now having problems reproducing this. Maybe network interface related?

lamping7, 2018-07-30:
I figured out how to reproduce this. Reopening as the question is still valid. I realize the question is more MAVLink related than the router, but here it goes the setup: mavlink-router is on the companion computer, connected to the serial UART port attached to the pixhawk's telem2 port. I'm running PX4. The router is meant to enable simultaneous connection via pymavlink and MAVROS. All of this works as it should, until I open up QGC on a ground computer, which connects to the pixhawk via 433MHz 3DR telemetry module, connecting from telem1 on the pixhawk and USB on the ground computer. 

So, does anyone have any insight as to why QGC interferes with this communication that prevents the script using pymavlink to receive the MISSION_REQUEST message, while other messages from PX4 are still able to be received?

lucasdemarchi, 2018-09-11:
Probably that mavlink-router itself doesn't know about QGC's sysid/compid pair. The flight stack seems to be sending this message on the wrong interface - if it's talking to QGC on a uart port, it should use that uart port only to talk to it.  If it sends over the uart that's going to the companion computer, then this happens.

lamping7, 2018-09-17:
So, this is a PX4 issue? I'm not sure what you mean. Does mavlink-router do something different with QGC traffic (I haven't looked at the implementation).

QiXuanWang, 2019-02-02:
I use mavproxy on companion computer connected to PX4 through telem2 port. I saw same information when setting target-system and target-component to 1, 1.
But after I add source-system and source-component as 255,1, those "unknown endpoints" messages are gone. 
I'm no expert on either but to log it here in case some update is happening.

floft, 2019-02-17:
I was running into similar issues. My companion computer would send messages that would be delivered until QGC (via 900MHz radio) downloaded its parameters, but after that the Pixhawk wouldn't get any of the sent messages. But, when I changed the system/component of my mavlink_connection as suggested by @QiXuanWang I no longer got these messages and my messages are delivered. I used system=1, component=2 for my companion computer (connected to telem2) and system=255 for the ground station (as is the default).

laurenweinstein1, 2020-03-29:
I'm helping someone with this issue (missing endpoints error on mission loads) and I must admit I do not understand this scenario. Exactly *where* are the system and component settings changed on an RPI running mavlink-router? Or are they changed in Mission Planner/QGC or somewhere else? Thanks!

laurenweinstein1, 2020-03-30:
So apparently there's no way without changing the code to alter these settings in mavlink-router -- however I see now that while the problem exists with Mission Planner it does not with QGC. So that's good enough for now, though understanding more of this issue in general would still be appreciated. Thanks.

MartinSollie, 2020-09-25:
I have been having the "unknown endpoints" messages occur when using ArduPilot and Mission Planner/MavProxy, simultaneously using
1. A SiK radio pair between MavProxy and a serial port on the autopilot (independent of mavlink-router)
2. UDP link from MavProxy to mavlink-router running on a companion computer, connected to the autopilot using a different serial port.

The event that caused the problem was TIMESYNC messages sent from Mission Planner using the SiK radio being forwarded by ArduPilot to the companion computer interface. Both the autopilot and the ground station would then appear as known components on the UART connection between mavlink-router and the autopilot. [This line](https://github.com/intel/mavlink-router/blob/8129480fb7143ddbd5e49667cc50adae6baa1823/src/mavlink-router/endpoint.cpp#L325) then prevents data sent fra Mission Planner over the UDP connection from being forwarded by mavlink-router to the autopilot.

The current master branch of ArduPilot has added the option to prevent mavlink forwarding on ports, but this is not yet in the stable releases.

jbeyerstedt, 2021-04-29:
Parallel communication links are an issue in mavlink-router because of the loop-prevention logic, as discussed in https://github.com/mavlink-router/mavlink-router/issues/240#issuecomment-776893101. But getting this right in all circumstances isn't very complex.

But the original issue is about the flight stack sending "response" messages for the ground station (e.g. after a parameter request) on all UART interfaces. But mavlink-router on the companion does not know to which endpoint these messages should be routed, because the ground station has never connected to mavlink-router. So discarding these messages is actually the right way in that case.

jbeyerstedt, 2021-09-09:
Closing this for now. It's no bug in mavlink-router and the case of the message was determined.

closed 2021-09-09 by jbeyerstedt