mavlink-router/mavlink-router #139: Support for multiple master links ?

davidbuzz, 2018-09-10:
In the event where there are two different types of uplink to a drone ( eg 900mhz , 2.4ghz , 4G(lte), 433mhz, GPS uplink etc ) , any two of these, or alternately two of of the same link type ( eg dual 900Mhz telem links), etc, then there should be some sort of failover, or redundancy, or support for using both link/s concurrently.   Mavproxy handles these sorts of scenario, aparently, but is clearly not as fast as mavlink-router.

lucasdemarchi, 2018-09-10:
do you mean a failover link that's only used when the first fails?

davidbuzz, 2018-09-10:
There are a number of redundancy scenarios, such as a master/slave failover, which is the minimum viable solution,  but a more generic solution would be one that allowed comms packets to arrive through either link, and would be aware of the active/inactive state for each link, and would send outgoing packets based on some smarts, e.g. send out whichever link was most recently received on, or assign a priority to the more capable link if more than one are online at the same time.

lucasdemarchi, 2018-09-11:
Just to explain a little bit what mavlink-router does: mavlink-router works by routing packets using pairs of (sysid, compid) and adding them to an endpoint. By doing that we don't care about much of the *content* of the message, only a few fields from the header. That's partially the reason why it's faster than mavproxy, since we don't care about parsing the entire message (mavproxy has a reason to do it that way).  And endpoint has no knowledge what's a "master". GCS, flight stack and a sensor over mavlink are all viewed as an endpoint.

So... we know if a (sysid, compid) is in more than one endpoint. Right now we just send the message to both channels (if the originator of the message is not the same link). In a flight stack it's the flight stack responsibility to respond on the right channel (if the message requires a response) or filter the message out since it already arrived on the other channel. Broadcast messages screw all this up, so there's a little bit more going on for this type of messages.

Is that filtering on the flight stack end not working for you or what? I think it should just work... How do I reproduce what you're doing?

davidbuzz, 2018-09-11:
What you've described "no knowledge of what is a master" seems to disagree with the README.md which says "There's one 'master' endpoint that should be the flight stack..."  Id like some clarity there, perhasps if the readme is incorrect, it needs more details there?  Now in our particular issue at present is that we have two (usb/ftdi) com ports, and the data can arrive basically through either of them, so we need to connect to more than one com port, and it doesn't appear to me from the readme that you can do that..?

lucasdemarchi, 2018-09-11:
Yeah, what's on the README may not have the best wording. Have you tried it and it failed?

davidbuzz, 2018-09-11:
Have I tried what and failed?  warning, sarcasm ahead. Are you suggesting I should randomly try every possible incantation that involves describing two comports, on the off chance I manage to stumble over an undocumented feature that might just possibly exist, but there is no evidence for?

lucasdemarchi, 2018-09-11:
I understood it was a bug report about something that should work that is not working.

davidbuzz, 2018-09-11:
This is an issue, that I had, and that I'm discussing.    At this stage it might be any of these, I don't know :
 
 - that the author/s never expected multiple masters to work, and it's considered out of scope.
 - that the author/s have not actively considered this scenario to this point, but think adding support for it is a good idea.
 - that this authors intend that the feature *should* already work, but it's not documented.

I don't know which if those is true, but it might be that if it's the second point, then this issue is a feature-request, or it may be that in the case of the third point, then this is a  documenation issue, which might be considered a bug report by some ( a documentation bug ).

lucasdemarchi, 2018-09-11:
As I mentioned above, it's the third: it should work. It needs testing though. I think this exactly what Emlid's Edge is doing adding multiple endpoints to their config file. If you want to take a look yourself, `examples/config.sample` is a good place. I can take a look on it later, but no ETA

lucasdemarchi, 2018-09-11:
Starting the flight stack:
```
$ ./build/sitl/bin/arducopter -M x
```

Starting mavlink-router:
```
./mavlink-routerd -c /tmp/main.conf
```
Mavproxy (or any other GCS supporting dual link):
```
$ mavproxy.py  --master udp:127.0.0.1:14550 --master udp:127.0.0.1:14551
```

The configuration file:
```ini
[General]
TcpServerPort = 11000

[UDPEndpoint bla-link1]
Address = 127.0.0.1
Mode = Normal

[UDPEndpoint bla-link2]
Address = 127.0.0.1
Mode = Normal

[TCPEndpoint sitl]
Port = 5760
Address = 127.0.0.1
Mode = Normal
```

For a stm32-based flight controller (rather than a linux/sitl) you will need to adapt the endpoints to use UARTEndpoint rather than TCPEndpoint

lucasdemarchi, 2018-09-20:
> I have reviewed https://github.com/intel/mavlink-router/blob/master/examples/config.sample and it contains only one uart, two udp, and one tcp connection, so it's not quite doing what you suggested.

yeah... It's just one example, not doing exactly what you wanted.

Did you have a chance to test?

RMIShane, 2019-08-23:
@lucasdemarchi

I have the same need that @davidbuzz was describing a year ago...

We have 900 and 4g coming from the drone simultaneously and need to receive both on the ground then choose which one to forward on to the GCS.

Did you determine if this is possible?

lucasdemarchi, 2019-08-23:
As I said above it should work. If it doesn't please post the logs.

lucasdemarchi, 2019-08-23:
The way it works is a little different from what you say. We send the message to both/multiple links and let the gcs/flightstack decide on what to drop based on msg id, sender, etc.

RMIShane, 2019-08-23:
Thanks for getting back with me!

A lot of GCSs don't have the capability of choosing so we are wanting to
build a base station that does that internally as well as provid a wireless
internet connection to the tablet or computer running the GCS.

How hard do you think it would be to implement a feature that would choose
between a primary and secondary incoming Mavlink connection to pass on to
the GCS?

Thanks,

Shane

RMIShane, 2019-08-27:
> The problem is how to define when or in what case to switch? It seems you
> want to choose one link, the one with fewer packet lost, to forward to a
> GCS. When implemented in a GCS it actually uses boths links. It may get one
> packet from one link and one from another. Then use the seq numbers to
> reconstruct the original ordering.
>
> It's possible to do any of these in mavlink-router, but not very trivial.
> PRs implementing it are welcome.

All we want to do is use the LTE link if available otherwise use RFD 900 so
we're not worried about any logic beyond that.

Would that make it any easier?

-Shane

lucasdemarchi, 2019-09-02:
So it seems more like supporting device hotplug than having any priority or adapt to multi-link. That sounds simpler at the surface but has some integration work to get it done. Right now we expect all endpoints to be created at startup rather than waiting for hotplug.

Supporting hotplug would be great as it means we don't have to order the startup of mavlink-router after devices are available in the system, making it better on supporting e.g. USB device that tend to show up later in the boot.

RMIShane, 2019-09-02:
The "priority" in the config would just avoid sending multiple heart beats
or redundant messages to the flight controller from the GCS even when you
have two or more active telemetry connections.  Basically the GCS endpoint
would only ever experience one Mavlink connection at a time as long as all
the other endpoints had a "priority" in the config file.

I wasn't aware whether you could "hot swap" or not but it does seem like a
feature that would make the overall system more robust.

-Shane

RMIShane, 2019-09-02:
I just realized you probably haven't gotten the direct email I sent yet.

I gave a detailed explanation of how this might be implemented and sent it
from my hybridproject email.

Hopefully that will make things more clear.

-Shane

jbeyerstedt, 2021-02-02:
Just as a follow up on this topic. Lucas is right, that at least QGroundControl (the only GCS I know) is fully capable of having multiple telemetry links and doing some failover between those.

A quite generic solution to your problem might be to have message de-duplication and endpoint groups in mavlink-router as proposed in #240. This would mean, that you can have multiple wireless links, which receive data from the drone, but only one instance of each packet will be forwarded to the GCS endpoint. In the other direction, all messages from the GCS to the drone will be sent via both wireless links.
De-duplication is not easy, but a time based failover can also be quite error-prone, if badly configured. But we need the features mentioned for a mesh network solution, so I would rather implement that. Do you think, this can solve your problem? Because then I would like to close this in favor of the other feature request.

RMIShane, 2021-02-02:
Anything is better than nothing :)

RMIShane, 2021-02-02:
I guess in all cases that's not true...

jbeyerstedt, 2023-02-18:
I would like to close this issue as:
1. Two (or even more) parallel links work when endpoint grouping is enabled (otherwise the "ground" side would reflect messages from one link back through the other one). And with de-duplication enabled as well, you would only see each message once.  
This has the lowest possible failover latency as each message is transmitted through both links, but your flight stack might respond to commands twice, if no mavlink-router is used on the UAV side to de-duplicate.
3. I don't think, that a true master-slave failover (only using one at a time) will be in scope of mavlink-router in the near future.

jbeyerstedt, 2023-02-20:
There are no questions in all of your four messages from 2021-02-02, just comments. If the proposal to have hot-swappable endpoints is still a thing, it should rather have its own issue/ feature request as this thread is already quite long. (Auterion started to implement some control interface to add new endpoints during runtime in their fork.)