mavlink-router/mavlink-router #140: Multiple sequential ULog invalid magic number failures

nslogan, 2018-09-11:
**System**

- `mavlink-routerd` running on OSD335x-SM (ARM A8)
- Connected via USB to PixHawk Cube running PX4 1.8.0

Configuration
```ini
# main.conf
[General]
Log=/var/log/flight-stack
LogMode=while_armed
MavlinkDialect=common
```

**Describe the bug**

I start mavlink-router: `mavlink-routerd /dev/ttyACM0:2000000`

The log output from mavlink-router is seen below:

[image]

**Expected Behavior**

I would expect that the log would start without issue.

**Additional context**

This behavior persisted through:

- Multiple arm/disarm cycles
- Multiple restarts of the flight controller and `mavlink-routerd`
- Changing link speed from 2000000 to 115200 

What additional information can I provide to help resolve this issue?

lucasdemarchi, 2018-09-11:
Try with 921600 baud rate. That's above the minimum for ulg and something that is known to work well for cube.

Also try without LogMode, maybe we had a regression since this is a new feature (btw, the option you're using got renamed).

nslogan, 2018-09-12:
Thank for getting back to me, Lucas.

I tried your suggestions, below is a breakdown of my results.

**First, I only changed the baud rate while leaving "LogMode=while-armed"**

[image]

I was able to log once successfully, but then failed the second time. I repeatedly armed/disarmed and found that *every other* attempt failed with 6-12 of these error messages.

**Then I left baudrate at 921600 and comment out my LogMode**

[image]

Interestingly, this showed the *same* behavior - every *other* attempt to start a log failed multiple times before succeeding.

**Additional Question**

Which option was renamed? Looking through `main.cpp` I see that the logging mode setting is `LogMode` and the value I want is `while-armed`, which matches my current configuration file.

lucasdemarchi, 2018-09-12:
@zehortigoza @bkueng you have more experience with the PX4 loggging. Any clue on this?

zehortigoza, 2018-09-12:
I would first check those 'x messages to unknown endpoints in the last 5 seconds':
Is QGC is getting all the information from PX4?
Enable mavlink-router debug messages and check what is the sysid/compid of this messages.

bkueng, 2018-09-13:
@nslogan There was a Firmware bug in 1.8.0 that could have triggered that. Can you try with master?
I also backported the fix to 1.8.0 in https://github.com/PX4/Firmware/pull/10475.

nslogan, 2018-09-13:
> > Which option was renamed?
> 
> in your original description you said `while_armed`, but the option is `while-armed`

I see, it looks like I cloned just before that commit and then checked for the name in GitHub's source browser without looking back at my own config file or first issue comment. Thanks for the clarification, I'll pull and update my build.

**Enable Debug Messasges**

Firmware: 1.8.0 (release; 96443b3)

> I would first check those 'x messages to unknown endpoints in the last 5 seconds':
> Is QGC is getting all the information from PX4?
> Enable mavlink-router debug messages and check what is the sysid/compid of this messages.

[image]

QGC is getting all of the information it needs from PX4, I'm able to interact with the vehicle normally.

Additionally, I left `mavlink-routerd` running longer this time and noticed it timing out on receiving log messages:

[image]

**Update to PX4 master**

Success! I re-enabled the "log while armed" setting and successfully armed / disarmed multiple times without any failures. So it appears the firmware was the issue.

I do, however, still have the same "<N> messages to unknown endpoints in the last 5 seconds" messages. Any ideas how I can fix that?

Thanks for the help, everyone.

zehortigoza, 2018-09-13:
> I do, however, still have the same " messages to unknown endpoints in the last 5 seconds" messages. Any ideas how I can fix that?

Enable mavlink-router debug messages and check what is the sysid/compid of this messages, then track who is sending that messages? if the destination should exist?

nslogan, 2018-09-13:
I captured a log from mavlink-router that contained the unidentified messages. All instances match this line:

```
Endpoint [4] got message to 1/0 from 1/1
	Known endpoints:
		1/1
Endpoint [-1] got message to 1/0 from 1/1
	Known endpoints:
		2/0
Message to unknown sysid/compid: 1/0
```

This appears to be a message from the flight controller, `MAV_COMP_ID_AUTOPILOT1`, to `MAV_COMP_ID_ALL`, which `mavlink-routerd` doesn't recognize as a valid endpoint. Is this at all related to issue #134. This appears to be the only message to the flight controller that mavlink-router is seeing, which is interesting.

zehortigoza, 2018-09-13:
So flight controller(sys id=1) is sending a message to sys id=1, so that is a issue in flight controller as it should not send that message externally. If you want to keep investigating this, add the message ID to the above message and check what PX4 module sends that ID.

nslogan, 2018-09-13:
Great, I can close out this issue then - looks like it was a PX4 firmware incompatibility issue.

Sorry, I'm not as familiar with the MAVLink environment yet. I'm not sure what you mean by:

> If you want to keep investigating this, add the message ID to the above message and check what PX4 module sends that ID.

I would like to track this down.

closed 2018-09-14 by nslogan