mavlink-router/mavlink-router #144: Autogen Failed on Raspi 3 B+

tizianofiorenzani, 2018-10-03:
I have already installed mavlink-router successfully on multiple raspberry Pi 3.

I am facing the following error when running ./autogen:

./configure: line 5897: syntax error near unexpected token `disable-static'
./configure: line 5897: `LT_INIT(disable-static pic-only)'

I installed autoconf package

tizianofiorenzani, 2018-10-03:
After running 

autoconf
aclocal

the ./autogen went through

closed 2018-10-03 by tizianofiorenzani

lucasdemarchi, 2018-10-03:
If you have the git repository cloned, this should always be sufficient to build from scratch:

```
git clean -ffxd
./autogen.sh c
make
```

If you are not moving the source code around from one board to another (or trying to use the .zip downloaded from github), then just the last command is sufficient.

tizianofiorenzani, 2018-10-04:
That actually worked great

tizianofiorenzani, 2018-12-01:
I got the same problem today with a fresh install. Tried everything, same issue

tizianofiorenzani, 2018-12-01:
Got clone

FLYBYME, 2019-02-12:
Fresh install on a pi3 `sudo apt install libtool` seems to fix it.