mavlink-router/mavlink-router #148: Mavlink-router still sends heartbeat even when disconnected from flight controller

reluctantgithubuser, 2018-11-22:
For this scenario:
Flight controller - companion computer (UART connection)

Companion computer running Mavlink-Router

Companion computer - GCS on ground computer (TCP/IP connection)

It seems that if the UART connection is broken, the mavlink-router still sends heartbeats to the ground computer from the companion computer.

This “tricks” the ground computer running the GCS into thinking the vehicle is still connected, even when it is not. This is a potentially very dangerous situation.

Is there a way to configure Mavlink-Router so that stops sending heartbeats to the ground computer from the companion computer if the vehicle disconnects from the companion computer?

lucasdemarchi, 2018-11-22:
It doesn't send heartbeat. How did you arrive to that conclusion?

reluctantgithubuser, 2018-11-22:
Ok. 

Mission planner does not disconnect when the UART connection goes down. In fact I can turn of the flight controller entirely and mission planner does not disconnect.

Furthermore in the terminal running mavlink-router it gives the following output "5 messages sent to unknown destination over last 5 seconds" (or something similar, I will check.) Every 5 seconds. So it is sending something.
I will run wireshark to see if I can figure out what it is sending.

Note: I actually have two computers running mavlink-router between the flight controller and GCS, so it is like this:

Flight controller -> companion computer -> cloud computer (Amazon) -> ground station computer
The companion computer and cloud computer both are running mavlink-router.

lucasdemarchi, 2018-11-22:
Which flight stack? Do you have logging enabled? That message is either the heartbeat from GCS to FC, not the other way around. Or the message we send to tell the FC to enable logging.

reluctantgithubuser, 2018-11-26:
Ok. I was wrong. mavlink-router does not send a heartbeat to the GCS. I analyzed the traffic with tcpdump and wireshark.

Here is what I found happens:

When the connection from the flight controller to mavlink-router on the companion computer is broken (e.g. the flight controller is powered down entirely):

No more heartbeats are sent to the GCS (Mission Planner on Win10 in my case).

However, GCS still sends heartbeats (one per second) to mavlink-router. 

Since the connection is over TCP, there is a TCP “ACK” to from the companion computer to the GCS for every heartbeat from the GCS to the companion computer.

But, no mavlink messages are ever sent to the GCS.

The question is, why doe the GCS still think it is still connected, even when it doesn’t receive heartbeats for a long time??? Is it because the TCP “ack” tricks the GCS into thinking it’s still connected all the way to the flight controller, when in reality it is only connected via TCP to the companion computer and mavlink-router?

Is there some setting in Mission Planner to make it say disconnected if it doesn’t receive a heartbeat from the flight controller after some timeout time?

lucasdemarchi, 2018-11-26:
It sounds like a bug in the GCS. 

> Is there some setting in Mission Planner to make it say disconnected if it doesn’t receive a heartbeat from the flight controller after some timeout time?

This will probably be answered better on their issue tracker

jbeyerstedt, 2021-02-10:
It seems to me, that this question is answered. Please comment, if I should re-open it.

closed 2021-02-10 by jbeyerstedt