mavlink-router/mavlink-router #150: Unable to parse config file on companion computer?

JacobCrabill, 2018-12-05:
Hi All,

I have mavlink-routerd running on a small arm-based SBC running OpenWrt, and it fails to parse the main.conf file.  Example error:

`On file /etc/mavlink-router/main.conf: Line 3: Invalid value '5760' for field 'TcpServerPort'`

If I leave that line commented out, it uses the default value of 5760 and runs fine.  This kind of problem occurs for the Port field of any TcpEndpoint I attempt to create.

Thanks,
Jacob

lucasdemarchi, 2018-12-05:
There are 2 different things: a `TcpEndpoint` is a TCP **client**. It will connect to whatever you configured in the Port setting.  TcpServerPort is a setting in the `General` section of the configuration file. See https://github.com/intel/mavlink-router/blob/master/examples/config.sample

JacobCrabill, 2018-12-06:
To clarify, here's a minimal example of a config file which will fail with the above error:

`[General]
TcpServerPort=5760
ReportStats=false
MavlinkDialect=ardupilotmega
`

JacobCrabill, 2018-12-06:
This seems to be a problem related to line endings - the `value_len` being assigned to the value for key `TcpServerPort` in conf_file.cpp covers the rest of the config file.  I broke out the `ConfFile::parse_ul()` function from the macro definition into a normal function, and added a print statement before the return to check the value being converted and returned, and this is what shows up:

Code:
```
int ConfFile::parse_ul(const char *val, size_t val_len, void *storage, size_t storage_len)
{
    char *str;
    assert(val);
    assert(storage);
    assert(val_len);
    if (storage_len < sizeof(unsigned long))
        return -ENOBUFS;
    str = strndupa(val, val_len);
    /// -- DEBUGGING
    unsigned long tmp;
    safe_atoul(str, &tmp);
    log_info("parsing ul %s as %s --> %ld\n",val,str,tmp);
    /// -- END DEBUGGING
    return safe_atoul(str, (unsigned long *)storage);
}
```
Result:
```
parsing ul 5760
ReportStats=false
MavlinkDialect=ardupilotmega
Log=/mnt/
#DebugLogLevel=info

# Use UART3 to connect to the autopilot
#[UartEndpoint pix]
#Device = /dev/ttymxc2
#Baud = 115200

 as 5760P_ --> 87264
```

Somehow the rest of the config file is being parsed as the value for TcpServerPort.  I'll keep looking into where/how value_len is getting set incorrectly.

lucasdemarchi, 2018-12-06:
Are you running master branch? Just tested here and it works for me:
```bash
$ cat /etc/mavlink-router/main.conf 
[General]
TcpServerPort=5761
ReportStats=false
MavlinkDialect=ardupilotmega
$ sudo ./mavlink-routerd 
Open TCP [4] 0.0.0.0:5761 *

#### in another terminal
$ sudo fuser 5761/tcp
5761/tcp:            13006
$ sudo cat /proc/13089/comm
mavlink-routerd

lucasdemarchi, 2018-12-06:
> This seems to be a problem related to line endings

may be...  we only ever tested with unix's line endings (\n), not dos/windows.

lucasdemarchi, 2018-12-06:
Even if I convert it to dos endings, it still works fine for me.

> Somehow the rest of the config file is being parsed as the value for TcpServerPort.

no, this is a problem in the way you printed. We don't modify the buffer to give to the parser, so we don't add `\0` to it. you need to print with something like:

```
     printf("%.*s\n", val_len, val);
```

What's strange in your case is str not being nul terminated.  That would be a bug in `strndupa()`. What libc are you using?

This works for me:
```
$ git diff 
diff --git a/src/common/conf_file.cpp b/src/common/conf_file.cpp
index d05cde0..271cb4e 100644
--- a/src/common/conf_file.cpp
+++ b/src/common/conf_file.cpp
@@ -516,6 +516,8 @@ int ConfFile::parse_bool(const char *val, size_t val_len, void *storage, size_t
             return -ENOBUFS;                                                    \
                                                                                 \
         str = strndupa(val, val_len);                                           \
+        log_info("parsing %.*s\n", (int)val_len, val);                          \
+        log_info("%s", str);                                                    \
         return _func(str, (_type *)storage);                                    \
     }
```

```
$ sudo ./mavlink-routerd -g debug
parsing 5761

5761
Open TCP [4] 0.0.0.0:5761 *
```

JacobCrabill, 2018-12-06:
Ah, right, the `str` variable has a shorter length so not directly the line endings - my bad on the printing.  These are unix line endings anyways though. 
Now that you mention libc though, I bet that's the problem.  I'm using basic build options in OpenWrt, which defaults to using musl (https://www.musl-libc.org/).  Any ideas on how to test / work-around strndupa if this indeed is the issue?

JacobCrabill, 2018-12-06:
Well, I fixed the problem just by directly copy-pasting my laptop's strndupa function into the parse_ul function:
```
    const char *__old = (val);
    size_t __len = strnlen (__old, (val_len));
    char *__new = (char *) __builtin_alloca(__len + 1);
    __new[__len] = '\0';
    (char *) memcpy (__new, __old, __len);

    str = __new;
```

Very strange that the strndupa that's getting compiled within the OpenWrt toolchain doesn't terminate the string properly.  I'm not even sure exactly where the string.h containing that function is getting included from during compilation.  Problem solved now on my end at least; not really a problem with mavlink-router after all.  I can just include a patch when compiling for OpenWrt in the future.

Thanks for your help!

closed 2018-12-06 by JacobCrabill

lucasdemarchi, 2018-12-06:
It would be nice to report the problem to musl... strndupa should terminate the string properly.

lucasdemarchi, 2018-12-06:
You could also do something like this in your end:

```diff
diff --git a/src/common/conf_file.cpp b/src/common/conf_file.cpp
index d05cde0..f7b746c 100644
--- a/src/common/conf_file.cpp
+++ b/src/common/conf_file.cpp
@@ -516,6 +516,7 @@ int ConfFile::parse_bool(const char *val, size_t val_len, void *storage, size_t
             return -ENOBUFS;                                                    \
                                                                                 \
         str = strndupa(val, val_len);                                           \
+        str[val_len] = '\0';                                                    \
         return _func(str, (_type *)storage);                                    \
     }
```

lucasdemarchi, 2018-12-06:
So... musl doesn't have strndupa. openwrt doesn't add it to musl neither. There is this repo that has a patch to musl: https://github.com/artynet/openwrt-patches/blob/master/patches-musl-1.1.16/903-string.h-add-strndupa.patch#L23

That implementation as well as the one you pasted here have a bug in the memcpy line. They should be copying `__len + 1`, not `__len`, otherwise it will not copy the terminating nul byte.

JacobCrabill, 2018-12-06:
Actually, as a follow-up, the strndupa problem was my own fault - and it's not (exactly) an issue with MUSL - because it's not a function included in MUSL's libc. I had added a definition of it some time ago in order to get mavlink-router to compile (and promptly forgot about it once it did compile). The implementation I copy-pasted from Google used strcpy without setting the final byte to '\0', hence the issue.

JacobCrabill, 2018-12-06:
Oops, just now saw your previous comment.
Yes, that's exactly where I copy-pasted from.