mavlink-router/mavlink-router #156: make: *** No rule to make target 'modules/mavlink/pymavlink/tools/mavgen.py', needed by 'include/mavlink/ardupilotmega/mavlink.h'.  Stop.

drnic, 2019-02-13:
Following the clone/submodule/build instructions, when I run `make` I get the following error/output. Protips for building?

I'm building inside an `ubuntu:bionic` docker image

```
# apt-get update
# apt-get install --no-install-recommends -y \
      dh-autoreconf pkg-config systemd \
      g++-6 gcc-6 \
      python-pip python-setuptools python-dev
# pip install pymavlink
# export CXX='g++-6'; export CC='gcc-6';
# ./autogen.sh && ./configure CFLAGS='-g -O2' \
             --sysconfdir=/etc --localstatedir=/var --libdir=/usr/lib64 \
           --prefix=/usr && \
         make && \
         make install

libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
libtoolize: linking file 'build-aux/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:13: installing 'build-aux/compile'
configure.ac:31: installing 'build-aux/config.guess'
configure.ac:31: installing 'build-aux/config.sub'
configure.ac:18: installing 'build-aux/install-sh'
configure.ac:18: installing 'build-aux/missing'
Makefile.am: installing 'build-aux/depcomp'
parallel-tests: installing 'build-aux/test-driver'

----------------------------------------------------------------
Initialized build system. For a common configuration please run:
----------------------------------------------------------------

./configure CFLAGS='-g -O2' CXXFLAGS='-g -O2' --sysconfdir=/etc --localstatedir=/var --libdir=/usr/lib          --with-rootprefix=         --with-rootlibdir=/lib

checking for gcc... gcc-6
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc-6 accepts -g... yes
checking for gcc-6 option to accept ISO C89... none needed
checking whether gcc-6 understands -c and -o together... yes
checking how to run the C preprocessor... gcc-6 -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... no
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking for style of include used by make... GNU
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking dependency style of gcc-6... gcc3
checking whether make supports nested variables... (cached) yes
checking for gcc-6 option to accept ISO C99... none needed
checking whether we are using the GNU C++ compiler... yes
checking whether g++-6 accepts -g... yes
checking dependency style of g++-6... gcc3
checking whether g++-6 supports C++11 features with -std=gnu++11... yes
checking for a Python interpreter with version >= 2.7... python
checking for python... /usr/bin/python
checking for python version... 2.7
checking for python platform... linux2
checking for python script directory... ${prefix}/lib/python2.7/dist-packages
checking for python extension module directory... ${exec_prefix}/lib/python2.7/dist-packages
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for fgrep... /bin/grep -F
checking for ld used by gcc-6... /usr/bin/x86_64-linux-gnu-ld
checking if the linker (/usr/bin/x86_64-linux-gnu-ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/x86_64-linux-gnu-ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc-6 object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc-6 supports -fno-rtti -fno-exceptions... no
checking for gcc-6 option to produce PIC... -fPIC -DPIC
checking if gcc-6 PIC flag -fPIC -DPIC works... yes
checking if gcc-6 static flag -static works... yes
checking if gcc-6 supports -c -o file.o... yes
checking if gcc-6 supports -c -o file.o... (cached) yes
checking whether the gcc-6 linker (/usr/bin/x86_64-linux-gnu-ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... no
checking how to run the C++ preprocessor... g++-6 -std=gnu++11 -E
checking for ld used by g++-6 -std=gnu++11... /usr/bin/x86_64-linux-gnu-ld -m elf_x86_64
checking if the linker (/usr/bin/x86_64-linux-gnu-ld -m elf_x86_64) is GNU ld... yes
checking whether the g++-6 -std=gnu++11 linker (/usr/bin/x86_64-linux-gnu-ld -m elf_x86_64) supports shared libraries... yes
checking for g++-6 -std=gnu++11 option to produce PIC... -fPIC -DPIC
checking if g++-6 -std=gnu++11 PIC flag -fPIC -DPIC works... yes
checking if g++-6 -std=gnu++11 static flag -static works... yes
checking if g++-6 -std=gnu++11 supports -c -o file.o... yes
checking if g++-6 -std=gnu++11 supports -c -o file.o... (cached) yes
checking whether the g++-6 -std=gnu++11 linker (/usr/bin/x86_64-linux-gnu-ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether _Static_assert() is supported... yes
checking whether _Noreturn is supported... yes
checking systemd system unit dir... /lib/systemd/system
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating config.h
config.status: executing depfiles commands
config.status: executing libtool commands

	mavlink-router 1
	================

	prefix:			/usr
	sysconfdir:		/etc
	libdir:			/usr/lib64
	rootlibdir:		/usr/lib64
	includedir:		${prefix}/include
	bindir:			${exec_prefix}/bin

	C compiler:		gcc-6
	C++ compiler:		g++-6 -std=gnu++11

make: *** No rule to make target 'modules/mavlink/pymavlink/tools/mavgen.py', needed by 'include/mavlink/ardupilotmega/mavlink.h'.  Stop.
```

lucasdemarchi, 2019-02-13:
did you run `git submodule update --init --recursive` as in the README?

drnic, 2019-02-14:
Yes; the module/mavlink/ directory is full.

lucasdemarchi, 2019-02-14:
post the output of  `git submodule  status --recursive`. You should have it like:

```
 13206d6f53aaff844f2d3595a01ac83a29e383db modules/gtest (release-1.8.0~17)
 bd73ce717013bfd94afd4b6eeeb29ed91875dc39 modules/mavlink (1.0.9-1946-gbd73ce7)
 54bf6e5a351b108b870690dee6763dd9d02756e0 modules/mavlink/pymavlink (54bf6e5)
```

drnic, 2019-02-14:
Ahh `--recursive` on the `git submodule`, sorry I missed that. Thanks a lot. Closing.

closed 2019-02-14 by drnic