mavlink-router/mavlink-router #158: compile failed on raspian

dk7xe, 2019-03-07:
Just tried to install mavlink router on another raspberry running raspian but `make` failed with

`~/src/mavlink-router $ make
  GEN      include/mavlink/ardupilotmega/mavlink.h
Traceback (most recent call last):
  File "./modules/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/pi/src/mavlink-router/modules/mavlink/pymavlink/generator/mavgen.py", line 12, in <module>
    from future import standard_library
ImportError: No module named future
Makefile:1609: die Regel für Ziel „include/mavlink/ardupilotmega/mavlink.h“ scheiterte
make: *** [include/mavlink/ardupilotmega/mavlink.h] Fehler 1`
 
**process steps**

1.  did a git clone
2. executed `git submodule update --init --recursive`
3. executed `./autogen.sh && ./configure CFLAGS='-g -O2' \
        --sysconfdir=/etc --localstatedir=/var --libdir=/usr/lib64 \
    --prefix=/usr`
4. executed `make`

lucasdemarchi, 2019-03-07:
Error message above: ImportError: No module named future

dk7xe, 2019-03-07:
Thanks for pointing me into the right direction.
`apt install python-future` solved the problem.

closed 2019-03-07 by dk7xe

sparklytopaz, 2020-01-09:
> Thanks for pointing me into the right direction.
> `apt install python-future` solved the problem.

I installed this but yet facing the same error

m1baldwin, 2020-01-17:
I have also seen this error, and not been able to fund a solution. I have also tried installing with 'pip install future' but this did not seem to help.

Jaeyoung-Lim, 2020-01-17:
@m1baldwin @sparklytopaz try `apt install python3 future`

m1baldwin, 2020-01-20:
Thanks @Jaeyoung-Lim , that seemed to solve the issue for me!