mavlink-router/mavlink-router #160: DTLS support

anandbaskaran, 2019-03-20:
Hello. Thanks for this tool. It is really helpful.

Do you plan to implement encryption support (perhaps DTLS)? This would be a really nice addon to this tool and could transport messages more securely.

Thanks in advance.

lucasdemarchi, 2019-03-20:
For IP it's a solved problem: make the transport use encryption.

Over radio, you need support on the flight stack side.

anandbaskaran, 2019-03-20:
Thanks for the quick response. Can you please point me to the documents/libraries to make the transport layer to use encryption.

Thanks in advance.

lucasdemarchi, 2019-03-20:
Can you clarify what you are trying to accomplish? Where are you routing the mavlink to/from? Is mavlink-router on the drone/ground/cloud?

anandbaskaran, 2019-03-20:
I am trying to use mavlink-router on the companion computer of the drone, that has connectivity to a remote server on the network. My goal is to make this communication as secure as possible.

jbeyerstedt, 2021-02-09:
I would agree with Lucas, that it's not part of the scope of mavlink-router to ensure transport-layer security. If you want to have a secured link, you should rather think about adding a VPN between your companion computer and server then. But you have to search by yourself, which VPN protocol is suitable for your application.