mavlink-router/mavlink-router #162: Dependencies Error at 'make' time on Raspberry Pi Zero W.

qkidamon, 2019-04-02:
Hi.

I'm in trouble while trying to build mavlink router on a up to date raspian system running in a Raspberry Pi Zero W.

First 3 steps (git clone, submodules and autogen) seem to be working fine. But when executing the 'make' step i get error messages related to failed dependencies into mavlink code. To be specific "mavgen" and "future" are missed.

I have installed "future" python module from raspian repository and error are still there.

I'm a bit lost with this and i really need it to work.

Thanks in advance.

Miguel.

Jaeyoung-Lim, 2019-04-02:
You can install future as the following

```
apt install python-future
```

qkidamon, 2019-04-03:
I have already done "sudo apt install python-future" as i said in my first post. Same error.

Should i do all build steps as root.

Jaeyoung-Lim, 2019-04-03:
Ah, have you tried installing with python3?

```
apt install python3-future
```
It might be a regression regarding the recent update to python3

qkidamon, 2019-04-05:
I've got it !!!

It builds OK (a few warnings regarding )

Thank you very much.

closed 2019-04-05 by qkidamon

NidalFayad, 2021-04-10:
Hello, kindly I couldn't install mavlink-router. I got this message after "sudo make install" how can I fix it?

make --no-print-directory install-am
 /usr/bin/mkdir -p '/usr/bin'
  /bin/bash ./libtool   --mode=install /usr/bin/install -c mavlink-routerd '/usr/bin'
libtool: install: /usr/bin/install -c mavlink-routerd /usr/bin/mavlink-routerd

lucasdemarchi, 2021-04-10:
@NidalFayad  this is not an error

NidalFayad, 2021-04-10:
@lucasdemarchi thank you. Yes I can communicate now mission planner with RPI  using this command "mavlink-routerd -e 192.168.1.3:14550" but I couldn't communicate in mavproxy "mavproxy.py --master=/dev/ttyAMA0 --baudrate 57600 --aircraft MyCopter --out 192.168.1.3:14550" 
I created config file like below but also didn't work:
[UartEndpoint rpi]
Device = /dev/ttyAMA0
Baud = 57600

Kindly how can I fix this problem?

lucasdemarchi, 2021-04-10:
Not sure what you are trying to do. Are you trying to use mavproxy together with mavlink-router? Please don't re-use unrelated closed issues like this.