mavlink-router/mavlink-router #163: Logging not working for MavId other than 1

anandbaskaran, 2019-04-10:
I am trying to log the mavlink messages of the drone with the following settings in the config file.

```
[General]
MavlinkDialect=auto
Log=/var/log/flight-stack
LogMode=always
DebugLogLevel=info
```

But this fails with the message `MAV_CMD_LOGGING_START result(2) is different than accepted`. Change back the `SYS_ID` to 1 resolves this problem.

lucasdemarchi, 2019-04-10:
ArduPilot or PX4?

ashwinsushil, 2020-02-11:
Hey @anandbaskaran @lucasdemarchi I have the exact same issue.
I am using PX4 and my configurations are:
```
[General]
TcpServerPort=5760
ReportStats=true
MavlinkDialect=common
Log=/var/lib/mavlink-router/
```
Let me know if you need additional information to close this issue.