mavlink-router/mavlink-router #164: Move maintenance to MAVLink org?

LorenzMeier, 2019-04-11:
Hi,

As the Intel team working on this has largely moved on to other open source efforts, I'd like to ensure the long-term viability of this component and would propose to move it to the MAVLink org, where we could maintain it as a reference implementation. Please think through it and let me know if that is welcome.

Otherwise we might start to maintain a fork with patches and "upstream" those, however, given there is no active development here that would be more out of respect for the OA and less practical.

lucasdemarchi, 2019-04-11:
This will take a while for approval. I already gave write privileges for @bkueng here. Why would you merge the patches on the auterion rather than merge them here? IMO it doesn't really matter where the repo are as long as the right people can merge stuff. Anyway, I'll see what can be done.

mrpollo, 2019-07-17:
Intel Team (cc @lucasdemarchi) any progress on discussions? there's been lots of movement lately on the MAVLink organization, the [bi-weekly dev call](https://mavlink.io/en/about/support.html#dev_call) keeps going strong ([meeting minutes](https://github.com/mavlink/mavlink/wiki)), there's a [healty ecosystem,](https://github.com/mavlink/rfcs/pull/15) the org recently gained the [MAVSDK](https://github.com/mavlink/mavsdk), and new content keeps getting generated / maintained, as you can see is the case of [docs](https://mavlink.io/en/), trainning content for [MAVSDK](https://www.youtube.com/watch?v=GXt-eXJ9vfg), and [advocacy for MAVLink](https://www.youtube.com/watch?v=-ftop8lvSPE) through developer events.

lucasdemarchi, 2019-07-19:
@mrpollo, no.

I don't doubt the value of mavlink org or the momentum gained with new projects. The repo here is still maintained by me and mavlink org is welcome to use and contribute as it is already happening. Question above is still answered: why does it matter where the repository is?

hamishwillee, 2019-08-01:
@lucasdemarchi 

A couple of reasons
- it will be easier to discover (by browsing) in MAVLink. It is the first thing you find by search on google, but I still discover a  lot of stuff by browsing. 
- It would be easier for users of the MAVLink org to invest in as a reference implementation.

If the project is being actively maintained and invested in by Intel (and you) then there isn't huge need to move it; And if having it in MAVLink org would make it harder for Intel to let you remain involved that would be a good reason not to move it. 

This isn't a power grab - we just want the most performant router in the MAVLink ecosystem to remain lively.

lucasdemarchi, 2020-11-14:
See #237. This was moved to github.com/mavlink-router

closed 2020-11-14 by lucasdemarchi