mavlink-router/mavlink-router #166: Mavlink Router Down on Arm

malfonsoNeoris, 2019-04-16:
Hi, i have a intel aero.
i have flashed ardupilot 3.6.7.

when i arm the copter, manually, via QGC or via python script.. the mavlink-routerd crash..

if i make a systemctl status mavlink_routered.sh
is sais core-dump... error status 11/SEGV

the only way to start the mavlink again is to turn off the copter, remove power, and power back
any help ?

lucasdemarchi, 2019-04-16:
Did you update the board to the latest version?

malfonsoNeoris, 2019-04-23:
hi lucas
heap i have 
Bios 1.016 (latest)
FPGA 0xc2 (latest)
OS ubuntu 16.04 kernel 4.4.76
FC firmware ardupilot 3.6.7 ( or 3.7dev.. .tested both)

currently... after ssh to the drone i start a mavproxy.py --master /dev/ttyS1 --baudrate 921600 --out XXXX to start a router server.. and work like a charm
it doesnt disconnect when armed by mavlink command/QGC/Manually

not the best, but a temporaly solution.
I am planning on maybe re install ubuntu... or back to yocto to try
As a comment... when i was testing PX4 firmware... never had this issue.

regards
MA

lucasdemarchi, 2019-04-23:
If you disable the logging, does it work?

malfonsoNeoris, 2019-04-23:
didnt tryed... but i will

lucasdemarchi, 2019-04-23:
There are 3 options for logging: enable on arm, always enable, always disable. From your description it seems we have a bug for the "enable on arm", because otherwise mavlink-router doesn't care if the vehicle is armed or not.

lucasdemarchi, 2019-04-25:
I tested locally and it worked fine. Are you using mavlink-router from master branch or the one that came installed?

closed 2020-03-17 by lucasdemarchi

malfonso0, 2020-03-17:
hi lucas.... sorry for this delay.
i forgot to update yo.. i disabled the log and worked.
then the proyect was cancelled and no further test.
thanks again for your responses.. sorry for not respond in time