mavlink-router/mavlink-router #179: ModuleNotFoundError: No module named 'future'

reluctantgithubuser, 2019-08-03:
Getting this error when running the make file:

ModuleNotFoundError: No module named 'future'

Hardware: Raspberry Pi Zero W
OS: 2019-07-10-raspbian-buster-lite

I have already run these commands:

sudo apt-get -y update                            
sudo apt-get -y upgrade
sudo apt-get -y install python  # 6 sec
sudo apt-get -y install python-wxgtk2.8 # 4 min  FAILS!!!!
sudo apt-get -y install python-matplotlib # 
sudo apt-get -y install python-opencv # 2 min
sudo apt-get -y install python-pip # 3 min
sudo apt-get -y install python-numpy  # 0 min
sudo apt-get -y install python-dev # 0 min
sudo apt-get -y install libxml2-dev # 1 min
sudo apt-get -y install libxslt-dev # 0.5 min
sudo apt-get -y install python-lxml # 0.75 min
sudo apt-get -y install python-setuptools # 0 min
sudo apt-get -y install git # 1 min
sudo apt-get -y install dh-autoreconf # 1 min
sudo apt-get -y install systemd # 0 min
sudo apt-get -y install nload # 0.5 min (network monitor, launch with nload -m)                                         
sudo apt-get -y install build-essential # 0 min

sudo pip install future
sudo pip install pymavlink
sudo pip install mavproxy

reluctantgithubuser, 2019-08-03:
This is the actual error message:

+ sudo make
  GEN      include/mavlink/ardupilotmega/mavlink.h
Traceback (most recent call last):
  File "./modules/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/pi/mavlink-router/modules/mavlink/pymavlink/generator/mavgen.py", line 12, in <module>
    from future import standard_library
ModuleNotFoundError: No module named 'future'
make: *** [Makefile:1664: include/mavlink/ardupilotmega/mavlink.h] Error 1

reluctantgithubuser, 2019-08-03:
Solved it-
You guys changed the make file to require python 3.
You should not do this so secretly just put it on the main page so people know.
Once using python 3, the build works now.

lucasdemarchi, 2019-08-04:
You either install python-future ( see the error message) or you use python3

closed 2019-08-04 by lucasdemarchi

reluctantgithubuser, 2019-08-04:
Thanks! I LOVE mavlink router!!!
Anyways, I did have the package future installed, but it still gave the error.
Switching to python3 worked, though.

guduzhe123, 2019-09-14:
I did have the package future installed, but it still gave the error.
How to change the make file to python 3?

sparklytopaz, 2020-01-09:
> I did have the package future installed, but it still gave the error.
> How to change the make file to python 3?

you use sudo apt install python3-future

dayjaby, 2021-02-15:
No intention of necro-bumping anything here, but with this command you can explicitly set the python version: `PYTHON=/usr/bin/python3 ./configure`

lucasdemarchi, 2021-02-15:
@dayjaby  can you create a PR mentioning this in the README?