mavlink-router/mavlink-router #185: Fail to built at RaspberryPie 3B+

yingshaoxo, 2019-10-07:
The OS is `2019-07-10-raspbian-buster-lite.iso`

The python version is  `Python 3.7.3 (default, Apr  3 2019, 05:39:12)`

The build command I was using is:
```
./autogen.sh && ./configure CFLAGS='-g -O2' \
        --sysconfdir=/etc --localstatedir=/var --libdir=/usr/lib64 \
    --prefix=/usr
```

The error I was getting is: 
```
checking for python extension module directory... ${exec_prefix}/lib/python2.7/dist-packages
./configure: line 5864: syntax error near unexpected token `disable-static'
./configure: line 5864: `LT_INIT(disable-static pic-only)'
```

chaserit123, 2019-10-09:
Did you ever figure this out?

lucasdemarchi, 2019-10-09:
Use python3 rather than python2.

And it's better to cross-compile than to use the poor rpi for compilation.

yingshaoxo, 2019-10-09:
> Did you ever figure this out?

No, in the end of the day, I just switched to `MAVproxy`. It can be interested by `pip3`, so easy to use.

The tutorial I made then is: https://medium.com/@yingshaoxo/px4-sitl-mavsdk-with-jmavsim-forward-udp-for-mavlink-to-work-with-jmavsim-and-raspberry-pi-197f62434203

mcelhennyi, 2019-10-30:
For anyone else:

`sudo apt-get install -y libtool
`

Got me past that point!

closed 2020-03-17 by lucasdemarchi