mavlink-router/mavlink-router #186: Route mavlink out another UART

chaserit123, 2019-10-09:
I've tried creating an output UART endpoint and having trouble getting it to work properly!!

lucasdemarchi, 2019-10-09:
What's the error?

chaserit123, 2019-10-09:
I haven't seen one but I'm not receiving mavling data on the destination devices UART. What's the proper way of defining another UART endpoint? Thank you

chaserit123, 2019-10-11:
Can someone provide an example of forwarding mavlink packets out another UART port?

pklapperich, 2019-10-23:
@chaserit123 what are you trying to connect to via the UART?

In my case I'm trying to bridge the PX4 simulator (which offers UDP connectivity) to a SiK radio where I'll have a GCS on the other end of the radio link. (IE, I'm trying to do px4 simulator (UDP) <-> mavlink-routerd <-> SiK radio <-> SiK radio <-> ground station)

As far as I can tell, the `UART device` and `UDP address` _that will be routed_ must be drones. I don't think you can connect a ground station on the other side of a UART device.

I think we need a new option `-u --uart-endpoint <uart> Add a UART endpoint client `. The `--help` output could also make it more clear that "endpoint client"s are GCS (QGroundControl, MissionPlanner, etc, etc) and connections "that will be routed" are drones.

lucasdemarchi, 2019-10-23:
It used to be the case we don't have a "master endpoint". There's no such thing in mavlink.  However there were some commits in the argument parser that broke this assumption. Doing a quick check, you should still be able to have multiple uart endpoints by using a configuration rather than command line args.  Take a look on examples/config.sample. Should be just a matter of adding multiple "[UartEndpoint your-name-here]

chaserit123, 2019-10-24:
I wanted to forward mavlink data to an OSD connected to my pi!

lucasdemarchi, 2019-10-24:
Just tested here with a simple configuration file using a fake serial port and it worked:

```
[General]
TcpServerPort=9000
ReportStats=false
MavlinkDialect=auto

[UartEndpoint radio]
Device = /dev/pts/3
Baud = 115200
```

Then I connected ArduPilot sitl on TCP:  `./build/sitl/bin/arduplane --model plane --uartA tcpclient:127.0.0.1:9000`

And mavproxy on the "fake" UART: `mavproxy.py --master /dev/pts/4`

in your case you probably want something like this:

```
[General]
TcpServerPort=9000
ReportStats=false
MavlinkDialect=auto

[UartEndpoint radio]
Device = /dev/ttyS1
Baud = 115200

[UartEndpoint fc]
Device = /dev/ttyS2
Baud = 115200

[UartEndpoint osd]
Device = /dev/ttyS3
Baud = 115200
```

Make sure to check the correct baud rate and serial device for your board.

chaserit123, 2019-10-25:
Thank you very much for this and BTW great job on this utility I absolutely love it!!

closed 2019-10-25 by lucasdemarchi