mavlink-router/mavlink-router #190: configuration error in Mavlink-router file

benittotomy, 2020-01-10:
pi@raspberrypi:~/mavlink-router $ ./autogen.sh && ./configure CFLAGS='-g -O2' --sysconfdir=/etc --localstatedir=/var --libdir=/usr/lib64 --prefix=/usr && make && make install
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
libtoolize: linking file 'build-aux/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:13: installing 'build-aux/compile'
configure.ac:18: installing 'build-aux/missing'
Makefile.am: installing 'build-aux/depcomp'

----------------------------------------------------------------
Initialized build system. For a common configuration please run:
----------------------------------------------------------------

./configure CFLAGS='-g -O2' CXXFLAGS='-g -O2' --sysconfdir=/etc --localstatedir=/var --libdir=/usr/lib          --with-rootprefix=         --with-rootlibdir=/lib          

checking for gcc... gcc-6
checking whether the C compiler works... no
configure: error: in `/home/pi/mavlink-router':
configure: error: C compiler cannot create executables
See `config.log' for more details

lucasdemarchi, 2020-03-17:
what does config.log say?

jbeyerstedt, 2021-07-28:
Closing because of inactivity. Please comment, if this is still an issue.

closed 2021-07-28 by jbeyerstedt