mavlink-router/mavlink-router #195: mavlink messages are rejected by autopilot unless I'm running receiver.py

dlwalter, 2020-02-08:
I am currently using mavlink-router to route from UDP on a RPI to a UART endpoint connected to the Autopilot.  I also have the loopback UDP as another endpoint (127.0.0.1:4000)

I am printing the received mavlink messages to the autopilot console.  

I can only receive the mavlink messages on the autopilot when I am running the example ./receiver.py to print the mavlink messages on the loopback endpoint.  If I stop receiver.py the autopilot rejects the mavlink messages.

lucasdemarchi, 2020-03-17:
could you post your configuration? what flight stack are you using?

dlwalter, 2020-03-17:
I'm running mavlink-router on a RPI CM3 connected over serial to a Pixhawk Cube Autopilot.

jbeyerstedt, 2021-02-09:
Sorry, for the late reply, but what's your mavlink-router configuration file and/or the command line parameters of mavlink-router?

jbeyerstedt, 2021-09-09:
Closing because of inactivity. Please comment, if I should re-open this question.

closed 2021-09-09 by jbeyerstedt