mavlink-router/mavlink-router #203: compiling on raspberry pi isn't smooth

olliw42, 2020-04-02:
I'm trying to install mavlink-router on a raspberry pi3 with a recent fresh raspbian (Raspbian Buster, Version: February 2020, Release date: 2020-02-13, Kernel version: 4.19), and it doesn't go smoothly.

I'm a unix illiterate largely and don't really know what the consequences are of what I am doing, but here are the deviations from the docs I did so far:

- ./autogen.sh did complain about autoreconf, so I did a sudo apt-get install autoconf which did get me past this

- ./configure xxxxx does complain about "syntax error near unexpected token disable-static, LT_INIT(disable-static pic-only)", so I have to edit the file configure.ac and uncomment the line  LT_INIT([disable-static pic-only]) (before running ./autogen.sh). This at least brings me past that complain, and the configure terminates with this mavlink router 2 overview info (I'm not sure I should post all the output)

btw: it's not clear what options to really chose for a "common configuration", the info in the README is seemingly different to the info outputted by ./autogen.sh. So I tried 

./configure CFLAGS='-g -O2' CXXFLAGS='-g -O2' --sysconfdir=/etc --localstatedir=/var --libdir=/usr/lib --prefix=/usr

and 

./configure CFLAGS='-g -O2' CXXFLAGS='-g -O2' --sysconfdir=/etc --localstatedir=/var --libdir=/usr/lib64 --prefix=/usr

- make does the compile but not the linking and stops with error

  CXXLD    mavlink-routerd
/usr/bin/ld: src/mavlink-router/logendpoint.o: in function `LogEndpoint::_fsync()':
/home/pi/Documents/Github/mavlink-router/src/mavlink-router/logendpoint.cpp:397: undefined reference to `aio_error64'
/usr/bin/ld: /home/pi/Documents/Github/mavlink-router/src/mavlink-router/logendpoint.cpp:404: undefined reference to `aio_fsync64'
/usr/bin/ld: src/mavlink-router/logendpoint.o: in function `LogEndpoint::LogEndpoint(char const*, char const*, LogMode, unsigned long, unsigned long)':
/home/pi/Documents/Github/mavlink-router/src/mavlink-router/logendpoint.cpp:62: undefined reference to `aio_init'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:880: mavlink-routerd] Error 1
make: *** [Makefile:696: all] Error 2

at this point I am lost now and can't find my way to get past this. Obviously something is wrong with <aio>, but I don't know exactly what to install (as usual with unix issues one fins 1000ends of "solutions" on the web, but all different and for different cases, so I can't determine which makes sense). sudo apt search libaio or sudo apt search aio appear to indicate a libaio is available.

I'm not sure if the first two attempts to resolve imply the third issue. But anyway, it's not smooth on raspberry currently.

any hint, help, suggestion would be much appreciated.
Thx

GoMartin300, 2020-04-02:
I've been able to get mavrouter working on RPi4, Raspbian buster following the instructions in this link: [https://discuss.px4.io/t/talking-to-a-px4-fmu-with-a-rpi-via-serial-nousb-ardunope/14119](url)
The author goes into installing mavsdk first, so scroll to the end of the explanation.
Please note that the author calls it a 'quick and dirty' method.

olliw42, 2020-04-02:
indeed !!! MANY THX
the key part I was missing is sudo apt install libtool 
I didn't do the sudo -s, but instead did sudo make install at the end, I find it easier

I'd like to mention that a main entry point to the instructions given here is the ArduPilot documentation page "Communicating with Raspberry Pi via MAVLink" (https://ardupilot.org/dev/docs/raspberry-pi-via-mavlink.html), which, by it's nature, might be a often used entry point, and also be used by not-so-experienced unix folks (like me). What I want to say is that I think that it would be kind if the instructions given here indeed would be such that they would work smoothly with a Raspberry Pi (or alternatively the AP docs to be updated to be more kind). I thus like to suggest to update the README accordingly.

For that reason I also want to summarize the steps I took (even though they are essentially identical to the excellent link just provided):

- install dependencies (may or may not be already present)(order is irrelevant)
  - sudo apt install python-future
  - sudo apt install python3-future
  - sudo apt install libtool
  - sudo apt install autoconf
- get mavlink-router files
  - cd into a folder where the git repo should be located (e.g. Documents/Github/)
  - git clone https://github.com/intel/mavlink-router.git
  - cd mavlink-router
  - git submodule update --init --recursive
- build&install
  - ./autogen.sh
  - ./configure CFLAGS='-g -O2' --sysconfdir=/etc --localstatedir=/var --libdir=/usr/lib --prefix=/usr
  - make
  - sudo make install

closed 2021-02-02 by jbeyerstedt