mavlink-router/mavlink-router #204: IPv6 support?

GoMartin300, 2020-04-02:
Is it possible to forward Mavlink traffic to a IPv6 endpoint?
I have it working now where mavrouter sends traffic to a public IPv4 (University Server), but would prefer not having to reserve an IPv4 and use IPv6 instead.

jbeyerstedt, 2020-09-01:
Just as a quick teaser: In a project I attend, we'll need to send to an IPv6 multicast address and receive multicast messages as well. So I'm currently working in 2 consecutive pull requests:
- General IPv6 support in mavlink-router
- Adding a new endpoint, which listens on a local port and sends to a pre-defined IP address (multicast or broadcast).

jbeyerstedt, 2020-11-12:
@GoMartin300 Could this be closed? (I forgot to add a "fixes #xyz" in the commit message.)

GoMartin300, 2020-11-12:
Judging by the documentation you provided I would say yes, this did not end up being a requirement for my project but I will definitely take advantage of this progress on the next one!

closed 2020-11-12 by lucasdemarchi