mavlink-router/mavlink-router #213: Error setting serial port latency, unable to reset UART afterwards, Jetson TX2

willdueease, 2020-05-22:
When running the router on a NVIDIA Jetson TX2 dev kit running stock jetpack, the router throws an error stating:

"Error while trying to write serial port latency: Unknown error -1"

the command is:
mavlink-routerd /dev/ttyTHS2 -e 192.168.0.50:49411

The port does appear to be set correctly, as the router works as intended after this. I can confirm with stty; the port is at 9600 at boot, and after running mavlinkrouter the port is now at 115200

That said, if I stop and restart the router, it fails to start, saying "could not reset uart"

in this state, stty says "input/output error" when I query the port. 

It does it same behavior on all the ttyTHS# ports, even if no serial device is attached to them.

I am able to configure and view the serial data using screen as many times as desired. 

I am attaching straces from the initial run, and the second failed run. 
[strace_second.5557.txt](https://github.com/intel/mavlink-router/files/4670090/strace_second.5557.txt)
[strace_first.5543.txt](https://github.com/intel/mavlink-router/files/4670091/strace_first.5543.txt)

Any help would be greatly appreciated, and am happy to run any troubleshooting tests as might be required.

lucasdemarchi, 2020-05-23:
```
ioctl(4, TCGETS, 0x7fe6d94620)          = -1 EIO (Input/output error)
```
Humn... since we are trying to *reset* the uart, maybe we could just write instead of querying the previous state.

GerardHale, 2020-06-05:
I have found the problem seems related to the fact that ioctl TIOCSSERIAL may not be supported by newer kernels. And once it is called on the NVIDIA TX2 the reset_uart function always fails until a reboot.  Disabling the entire section in UartEndpoint::open() marked with 
    // For Linux, set high speed polling at the chip
    // level. Since this routine relies on a USB latency
    // change at the chip level it may fail on certain
    // chip sets if their driver does not support this
    // configuration request
fixes this
I could not find a modern way using 
tcgetattr/tcsetattr to fix this
It looks like a place for a conditional compile

lucasdemarchi, 2020-06-06:
> I have found the problem seems related to the fact that ioctl TIOCSSERIAL may not be supported by newer kernels.

I think that rather than "new kernels", the problem is more that not every serial is behind a USB bus, not every tty driver in the Linux kernel are forced to implement this method (see drivers/tty/tty_io.c:tty_tiocsserial) and that you may be hitting a bug on a particular driver.

ACPAviation, 2020-10-18:
Is this the same issue?
Ubuntu 18.04.5 LTS Kernal 5.4.0-51

Could not reset uart 
mavlink-routerd -e 192.168.7.1:14550 -e 127.0.0.1:14550 /dev/ttyACM0:115200

GerardHale, 2020-10-19:
Quuite probably it seems the authors expect a usb serial reguardless of the device used. I suggest a switch on device name 
ttySnn vs ttyUSBnn to enable the code

lucasdemarchi, 2020-10-19:
@GerardHale / @ACPAviation it gives you a warning if it can't set it. But it should continue to work regardless of that. I checked the code and it seems to be the case.

TomRvr, 2021-01-07:
I faced the same issue  
It seems that the `reset_uart` is doing something wrong when closing the connection  
I fixed it by commenting this line in UartEndpoint destructor : https://github.com/mavlink-router/mavlink-router/blob/51c72ffdce5ffb9d3f07010cd00de9dca4f8af4c/src/mavlink-router/endpoint.cpp#L491

PR #409 (2023-10-13): Don't perform UART reset on shutdown
This fixes the issue exhibited in #213. I have tested this on a Jetson TX2 and confirmed that by not performing the reset, the UART port is still accessible with minicom after shutting down mavlink router. For what its worth, setting stty's sane on the UART does not cause the same problem, so there is evidently some difference between the `reset_uart`  and stty sane despite the comment in the code. 

I hate to be the guy who offers a solution that's just "delete this code and it works", but its not clear to me what the underlying issue is and this does fix  [... cut]
closed 2023-10-13 by lucasdemarchi