mavlink-router/mavlink-router #214: TCP server rejects connections after first closes

highfreq, 2020-05-28:
I can connect to TCP server on port 5790 ( i changed port ) with no problem the first time with pymavlink from my application. Once i close my application i can see mavlink-router closing the connection in status of the service. If i try to connect again than it doesn't connect anymore unless i restart mavlink-router service.
This is the error i get from pymavlink:
ConnectionRefusedError: [WinError 10061] Impossible to establish a connection. Persistent decline from remote computer

highfreq, 2020-05-28:
Ok, investigated a bit more, the service closes once i close my application, that is why it doesn't answer on second try. I solved it by changing the following in service file:
Restart=Always
commented restart time
Now it restarts immediately after i close my connection and works on second attempt.

lucasdemarchi, 2020-05-29:
Do you mean mavlink-router stops after you close your app?

Jaeyoung-Lim, 2020-05-29:
@highfreq For me this is not the case, I can connect / reconnect multiple times from QGC

highfreq, 2020-05-29:
Yes it closes if i open/close my app or MP multiple times.
I solved it by restart=always

highfreq, 2020-06-02:
Don't know if it was clear in my previous comments but mavlink-router closes only if i use connections to the tcp server. If i use udp it doesn't happen.

agrueing, 2020-07-06:
This one can probably be closed, resolved by https://github.com/intel/mavlink-router/commit/555bd0150f1043387f4b158f586bda504c6ee038

Found this issue investigating similar behavior when killing apmplanner2 while params were still downloading, turns out the version I was using was built from master around the beginning of May.

closed 2021-02-02 by jbeyerstedt