mavlink-router/mavlink-router #215: routing_test.py fails

JonasVautherin, 2020-05-31:
On latest `master`, with Ubuntu 18.04, when I run:

```
 python3 tests/routing_test.py
```

I get:

```
Could not open conf file '/nonexistent' (No such file or directory)
Open UDP [4] 127.0.0.1:10000 *
Open UDP [5] 127.0.0.1:10101  
Open UDP [6] 127.0.0.1:10100  
Open TCP [7] 0.0.0.0:5760 *
10 messages to unknown endpoints in the last 5 seconds
sender100 expected 9 messages, got 0

receiver100 expected 19 messages, got 10

Routing test FAILED. See previous output for more information
```

What could be happening here?

JonasVautherin, 2020-05-31:
I see that Travis fails, too: https://github.com/intel/mavlink-router/pull/216

lucasdemarchi, 2020-06-04:
Known failure. We changed routing rules.

commit 7086ed67 (2021-11-30): Fix routing test: set component ID properly (fixes #215)
closed 2021-11-30 by lucasdemarchi