mavlink-router/mavlink-router #220: Getting mavlink feed from TCP but not UDP on QGC.

MaximeAubanel, 2020-06-09:
On my raspberry Pi I launched the router as follow:

```
mavlink-routerd -e 0.0.0.0:14550 /dev/ttyPixhawk:921600
```

It displays

```
Could not open conf file '/etc/mavlink-router/main.conf' (No such file or directory)
Error while trying to write serial port latency: Unknown error -1
Open UART [4] /dev/ttyPixhawk *
UART [4] speed = 921600
Open UDP [5] 0.0.0.0:14550  
Open TCP [6] 0.0.0.0:5760 *
```

If I add a comm link on QGC over TCP using my RaspberryPi IP and port 5760 it works.

But if I add a comm link on QGC over UDP using my RaspberryPi IP and port 14550 it doesn't work.

I notice that there is a difference between `0.0.0.0:14550` and `0.0.0.0:5760 *`. The wildcard isn't on my rule. I tried to add it but it doesn't work.

MaximeAubanel, 2020-06-11:
Same thing when connecting with MAVROS.

jbeyerstedt, 2020-09-01:
You need to specify the target IP address on a UDP endpoint, if it's in normal mode. When QGC is running on the same machine, use the localhost/ loopback address 127.0.0.1. In your case the UDP packets don't know where to go, because no network interface has 0.0.0.0 as address. (You can only use that one to receive from everyone, when opening a port with mavlink-router.)

And the asterisk in the log output shows, that mavlink-router will bind to/ listen on that IP and port. It's no wildcard. I also had to search the source code to find that piece of information.

jbeyerstedt, 2021-02-02:
I'll close this now because of inactivity. Please comment, if this issue is still unsolved.

closed 2021-02-02 by jbeyerstedt